ebroecker/canmatrix #3: Converting a dbc file with chinese characters aren't working

nexulm, 2015-11-05:
18 Frames found
Traceback (most recent call last):
  File "convert.py", line 71, in <module>
    db = ex.exportXls(db, outfile)
  File "/cygdrive/v/canmatrix/library/exportxls.py", line 288, in exportXls
    writeFrame(frame, worksheet, row, framestyle)
  File "/cygdrive/v/canmatrix/library/exportxls.py", line 49, in writeFrame
    worksheet.write(row, 0, label = "%3Xh" % frame._Id, style=mystyle)
  File "/usr/lib/python2.7/site-packages/xlwt/Worksheet.py", line 1088, in write
    self.row(r).write(c, label, style)
  File "/usr/lib/python2.7/site-packages/xlwt/Worksheet.py", line 1136, in row
    self.__rows[indx] = self.Row(indx, self)
  File "/usr/lib/python2.7/site-packages/xlwt/Row.py", line 43, in __init__
    raise ValueError("row index was %r, not allowed by .xls format" % rowx)
ValueError: row index was 65536, not allowed by .xls format

ebroecker, 2015-11-05:
Could you provide some example-dbc for me? 
I think I used UTF8, but if we have problems with chinese characters, there seems to be an error.

Best would be a small .dbc that raises this error.

nexulm, 2015-11-06:
Sorry I don't have a small dbc file which raises this error. Attached is a file (*.jpg => *.dbc) to demonstrate that the chinese character within the DBC file is not correctly translated to xls format.

By the way: What is regarding supporting Excel 2010? The xlsx format has some improvements regarding row count (>65536). This was the last line from the error log above although the input DBC file includes only 137 CAN-Signals!

[image]

ebroecker, 2015-11-06:
first attempt: please modify line 36 of importdbc.py to following: "dbcImportEncoding = 'utf-8'"
Afterwards I can see chinese characters in the excelsheet.
This is a old problem of the dbc-files. 
I never found some hint which charset it uses.
Maybe I need to add some commandline option for the charset which should be used while importing dbc.

nexulm, 2015-11-06:
For the single CAN-signal your proposed change works but the complete file generate now another kind of error which is utf-8 related!?!
Traceback (most recent call last):
  File "convert.py", line 39, in <module>
    dbs[""] = im.importDbc(infile)
  File "/cygdrive/v/canmatrix/library/importdbc.py", line 82, in importDbc
    db._fl.addSignalToLastFrame(Signal(temp.group(1), temp.group(2), temp.group(3), temp.group(4), temp.group(5), temp.group(6), temp.group(7),temp.group(8),temp.group(9),temp.group(10).decode(dbcImportEncoding),reciever))
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xa8 in position 0: invalid start byte

ebroecker, 2015-11-06:
hmm - maybe another charset?

just uploaded a first xlsx-export. Just for more rows support.

ebroecker, 2015-11-07:
Well - it seems you have utf8 comments for your "Signal-Coments", but something like iso-8859-1 for "Signal-Descriptions".
I have to think about, how to tell the import-module different encodings for different entities in the dbc by commandline.

I'll find a solution, stay tuned

By the way - I guess it is not possible to get the dbc for testing purpose only? 
If it would be possible - just send it to: eduard at gmx dot net

ebroecker, 2015-11-07:
please try newest version with following command line argument: --dbcCommentCharset=utf8

Example: ./convert.py --dbcCommentCharset=utf8 input.dbc output.xlsx

nexulm, 2015-11-08:
My sample dbc is working with the new mentioned parameter but the Original chinese file throws this error:
Traceback (most recent call last):
  File "convert.py", line 59, in <module>
    dbs[""] = im.importDbc(infile, cmdlineOptions.dbcCharset,  cmdli             neOptions.dbcCommentCharset)
  File "/cygdrive/v/canmatrix/library/importdbc.py", line 108, in im             portDbc
    signal.addComment(temp.group(3).decode(dbcCommentEncoding).repla             ce('\\"','"'))
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xb2 in position              2: invalid start byte

ebroecker, 2015-11-09:
I added some 'try'-'execpt'-Blocks (only for comment-decoding).
Thus convert.py should run through with your dbc while throwing some errors.

Maybe this helps me to find out more about your dbc.

nexulm, 2015-11-09:
I was able to convert the DBC now to xlsx by using ./convert.py --dbcCommentCharset=GB2312 input.dbc output.xlsx. My Notepad++ shows the simplified chinese charset GB2312 for the opened dbc file.
The only thing what is still not correct is the conversion of combined comments like "OOPS钥匙认证成功" (dbc opened in Notepad++) and "OOPSÔ¿³×ÈÏÖ¤³É¹¦" (xlsx opened in Excel 2010)

ebroecker, 2015-11-09:
did you try ./convert.py --dbcCommentCharset=GB2312 --dbcCharset=GB2312 input.dbc ?

nexulm, 2015-11-09:
Oops, no. After using ./convert.py --dbcCommentCharset=GB2312 --dbcCharset=GB2312 input .dbc output .xlsx I get this error:
Traceback (most recent call last):
  File "convert.py", line 59, in <module>
    dbs[""] = im.importDbc(infile, cmdlineOptions.dbcCharset,  cmdlineOptions.dbcCommentCharset)
  File "/cygdrive/v/canmatrix/library/importdbc.py", line 90, in importDbc
    db._fl.addSignalToLastFrame(Signal(temp.group(1), temp.group(2), temp.group(3), temp.group(4), temp.group(5), temp.group(6), temp.group(7),temp.group(8),temp.group(9),temp.group(10).decode(dbcImportEncoding),reciever))
UnicodeDecodeError: 'gb2312' codec can't decode bytes in position 0-1: illegal multibyte sequence

nexulm, 2015-11-09:
./convert.py --dbcCommentCharset=GB2312 --dbcCharset=gbk input .dbc output .xlsx does the trick for me.
But I'm not able to convert back from the .xlsx to .dbc
Traceback (most recent call last):
  File "convert.py", line 65, in <module>
    dbs["default"] = im.importXls(infile)
  File "/cygdrive/v/canmatrix/library/importxls.py", line 34, in importXls
    wb = xlrd.open_workbook(filename, formatting_info=True)
  File "/cygdrive/v/canmatrix/xlrd/__init__.py", line 422, in open_workbook
    ragged_rows=ragged_rows,
  File "/cygdrive/v/canmatrix/xlrd/xlsx.py", line 751, in open_workbook_2007_xml
    raise NotImplementedError("formatting_info=True not yet implemented")
NotImplementedError: formatting_info=True not yet implemented

ebroecker, 2015-11-09:
sorry - no xlsx-read support yet (this may be wrong documentated). 
Xlsx-readsupport is to be implemented first. 
I'll close this issue and create a new one for xlsx-read-support.

closed 2015-11-09 by ebroecker