ebroecker/canmatrix #4: xlsx-read support

ebroecker, 2015-11-09:
we have currently  no support for reading xlsx-matrixes

nexulm, 2015-11-09:
Is a timing for implementing reading xlsx files already available?
The special dbc with chinese characters creates more than 65535 rows so that the xlsx format must be used. But currently we don't have any way back from a modified .xlsx to .dbc with the canmatrix convert tool. :-(

ebroecker, 2015-11-09:
should work.

BUT be aware, exporting to excel looses Information.
Doing roundtrip (dbc->xls(x)->dbc) you will a lot of extra attributes, which are possible in dbc.

nexulm, 2015-11-09:
xlsx read works now, also with my chineses charset file.
But I've seen the issues as follows:
1) units are missing after dbc => xlsx => dbc conversion
2) scaling of i.e. min/max values for float values is not correct
3) some signed/unsigned values have new min/max values after dbc => xlsx => dbc conversion

ebroecker, 2015-11-10:
I'm gonna have a look on it.

ebroecker, 2016-08-30:
Should be Fixed in branch stableApi with #b0a0389

xlrd also Supports xlsx-files, so xlsx Importeur is deprecated now

closed 2016-08-30 by ebroecker