ebroecker/canmatrix #5: converting dbc <=> xls <=> dbc

nexulm, 2015-11-09:
converting a dbc to xls and vice versa back to dbc again without changing anything I get this fault:

Traceback (most recent call last):
  File "convert.py", line 65, in <module>
    dbs["default"] = im.importXls(infile)
  File "/cygdrive/v/CANtools/canmatrix/library/importxls.py", line 217, in importXls
    newSig._factor = float(factor)
ValueError: could not convert string to float: degree

ebroecker, 2015-11-09:
found a small bug, please try xls->dbc again...

nexulm, 2015-11-09:
OK, it's working now (again)!
I'm currently confused that the output dbc file has "default_" as prefix (=> "default_output.dbc").

ebroecker, 2015-11-09:
OK, thanks for testing and finding bugs :-)

yes the "default" is confusing. 
This is because .arxml can contain multiple can-configurations and the importers all use the same api.

ebroecker, 2015-11-09:
fixed 'default'-issue

closed 2015-11-09 by ebroecker