ebroecker/canmatrix #12: Python3 support

altendky, 2015-11-28:
I'm working on Python3 support and needed somewhere to put a few questions.

@ebroecker, while looking at changes in the test result files while adjusting for Python3 I noticed that some of the .yaml test reference files have [`!!python/unicode` in front of the strings](https://github.com/ebroecker/canmatrix/blob/77004631286696cf948cb9865ef5881d7d9f1827/test/reference/from_dbc/test.yaml#L15).  I noticed because the Python3 stuff wasn't writing that.  I'm guessing we don't want references to Python in our YAML files but wanted to confirm with you.

I also had issues where some of the items were being output in different orders than before which is annoying for running the diff as the test.  Specifically, [some of the def's here](https://github.com/ebroecker/canmatrix/blob/77004631286696cf948cb9865ef5881d7d9f1827/test/reference/from_arxml/CAN_test.dbc#L224).  Am I correct that these would be OK in any order so I can sort the list prior to writing it to the file?

@nexulm, I am trying to retain the dual-encoding option Eduard added for you but it would be really helpful if I could get the reference file.  Since we have a basic regression test script now it would also be nice to be able to include such a reference file in the tests.  My email is in my profile, or we can arrange another transfer method.

ebroecker, 2015-11-28:
yaml: this format is not used by anybody, this is just for dumping a database.  I saw these !!python/unicode references. I have no idea how to avoid this with python2. 

Yes - the ordering issue is annoying. Lets implement some sorting.

altendky, 2015-11-30:
Preliminary work in 899655e375e3bfe94eb8db3d8a21a0aa0ea0c987.  Expect to attempt to return 2 support before calling this complete.

altendky, 2015-11-30:
Easier than expected to get 2 support back.  See 1b2b4d2721476762b8f956d5c0acf1fe73ace687.  YAML !!python artifacts remain for now in Python 2.

ebroecker, 2015-12-03:
python3 works now

closed 2015-12-03 by ebroecker