ebroecker/canmatrix #14: Error message converting dbc to Excel w and w/o --dbcCharset=utf8

nexulm, 2015-11-30:
With latest Github sync I'm not able to convert dbc to xls/xlsx anymore!

Traceback (most recent call last):
  File "convert.py", line 122, in <module>
    sys.exit(main())
  File "convert.py", line 119, in main
    charset=args.dbcCharset, comment_charset=args.commentCharset)
AttributeError: 'list' object has no attribute 'dbcCharset'

nexulm, 2015-11-30:
After latest bugfix #14 the commentCharset is missing!

Traceback (most recent call last):
  File "convert.py", line 122, in <module>
    sys.exit(main())
  File "convert.py", line 119, in main
    dbcCharset=cmdlineOptions.dbcCharset, dbcCommentCharset=cmdlineOptions.commentCharset)
AttributeError: Values instance has no attribute 'commentCharset'

ebroecker, 2015-11-30:
Thanks again --- I learned once again: never change code without testing !!!

commit a848341f (2015-11-30): bugifx

should fix #14 - untested
closed 2015-11-30 by ebroecker