ebroecker/canmatrix #15: dbc export error

ebroecker, 2015-11-30:
while exporting dbc I get following error. 
Seems to be some type missmatch 

 File "convert.py", line 68, in convert
    ex.exportDbc(db, outfile, dbcCharset,  dbcCommentCharset)
  File "somepath\canmatrix\library\exportdbc.py", line 129, in exportDb
c
    for (type,define) in sorted(list(db._globalDefines.items()), key=lambda x: i
nt(x[0])):
  File "somepath\canmatrix\library\exportdbc.py", line 129, in <lambda>

    for (type,define) in sorted(list(db._globalDefines.items()), key=lambda x: i
nt(x[0])):
  File "c:\Python27\lib\site-packages\future\types\newint.py", line 76, in __new
__
    return super(newint, cls).__new__(cls, val)
ValueError: invalid literal for long() with base 10: 'BusType'

altendky, 2015-11-30:
I think there are lots of opportunities for cleanup of things like this, cleanly handling various types and formats, but I am personally ok with your quick fix for now.  The tests still 'passed' for me.  They fail, but only on the .xls (2 and 3) and .yaml files (2).

I would suggest that we could add other files, or edit the existing ones, to make sure these scenarios are getting tested as we move forward.

altendky, 2015-12-01:
Should have commented here, not on the commit...

The point of that was to get natural sorting. In at least some cases I would get lists like 1,10,11,12,2,3... I think we should likely convert to int on import for numeric fields.

closed 2015-12-03 by ebroecker