ebroecker/canmatrix #16: dbc import error (endswith() does not work with str)

m8ddin, 2015-11-30:
```
File "convert.py", line 35, in convert
    dbs[""] = im.importDbc(infile, dbcCharset,  dbcCommentCharset)
  File [...]/canmatrix/library/importdbc.py", line 80, in importDbc
    if l.endswith('";'):

TypeError: endswith first arg must be bytes or a tuple of bytes, not str
```

Possible solution: conversion to bytestring?
``` if l.endswith('";'): -->  if l.endswith(b'";'):```
(in lines 59, 69, 79)

ebroecker, 2015-11-30:
This seems to be a python3-issue; python3 is not yet supported - but we are working on it...

c:\Python34\python.exe convert.py someTest.dbc aa.dbc

Importing someTest.dbc ...
Traceback (most recent call last):
  File "convert.py", line 122, in <module>
    sys.exit(main())
  File "convert.py", line 119, in main
    dbcCharset=cmdlineOptions.dbcCharset, dbcCommentCharset=cmdlineOptions.dbcCo
mmentCharset)
  File "convert.py", line 35, in convert
    dbs[""] = im.importDbc(infile, dbcCharset,  dbcCommentCharset)
  File "C:\Users\broecked\canmatrix\library\importdbc.py", line 79, in importDbc

    if l.endswith('";'):
TypeError: endswith first arg must be bytes or a tuple of bytes, not str

---------------

C:\Users\broecked\canmatrix>c:\Python27\python.exe convert.py someTest.dbc aa.dbc
Importing someTest.dbc ...
done

Exporting aa.dbc ...

203 Frames found
done

altendky, 2015-11-30:
I think you're plan of just making the string literals bytestrings is probably good.  What I did was open the files in binary so I could control the encoding in and out.  As such, all the regexes had to be converted to bytes and the other literals probably need it as well.  I think there are other approaches than my fairly brute force one but for the moment I will review the remaining string literals and try to correct them.

A test case which triggers this would also be appreciated.

altendky, 2015-11-30:
e9b6c96179b4a3030f78461b4384ba4a27ce854a

Changed per your suggestion.

m8ddin, 2015-11-30:
You're right, it is a Python3 issue. Using bytestring was my 'quickfix', should be tested thoroughly.

commit e9b6c961 (2015-12-01): Correct importdbc string literals to byte strings per #16

After the 2 to 3 conversion these literals were missed despite my editor
highlighting them for me.  Hopefully this fixes #16 and a test case can
be provided to avoid regression.
closed 2015-12-01 by altendky