ebroecker/canmatrix #26: Export I-SIGNAL-TO-I-PDU-MAPPING

jmtatsch, 2016-01-11:
Hi,

I am attempting to convert a large arxml file to dbc, but no dbc file is created although  524 I-SIGNAL-TO-I-PDU-MAPPINGs are found.

```
INFO - convert - Importing my.arxml ... 
DEBUG - importarxml - Read arxml ...
DEBUG - importarxml -  Done

DEBUG - importarxml - Build arTree ...
DEBUG - importarxml -  Done

DEBUG - importarxml - DEBUG 0 frames in arxml...
DEBUG - importarxml - DEBUG 0 can-frame-triggering in arxml...
DEBUG - importarxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - importarxml - DEBUG 524 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
INFO - convert - done

INFO - convert - Exporting my.dbc ... 
INFO - convert - done
```

Is that to be expected? How can I get a clean list of signals?

ebroecker, 2016-01-11:
currently we need the TAG "SIGNAL-TO-PDU-MAPPINGS" in the arxml. This is the container which connects the can-IDs with the PDUs. 
Could you provide your arxml? Maybe to my private email given in my profile?
Ist it (again) a GM-arxml (compare issue #23 )?
Eduard

jmtatsch, 2016-01-12:
Dear Eduard,

unfortunately, i cant share the whole arxml file with you. I could edit out and redact certain parts for you if necessary. 
There is not a single tag <SIGNAL-TO-PDU-MAPPINGS> only <I-SIGNAL-TO-PDU-MAPPINGS> tags.

```
                <I-SIGNAL-I-PDU>
                  <SHORT-NAME>FrontCamera</SHORT-NAME>
                  <LENGTH>1146</LENGTH>
                  <I-PDU-TIMING-SPECIFICATIONS>
                    <I-PDU-TIMING>
                      <MINIMUM-DELAY>0.045</MINIMUM-DELAY>
                      <TRANSMISSION-MODE-DECLARATION>
                        <TRANSMISSION-MODE-TRUE-TIMING>
                          <EVENT-CONTROLLED-TIMING>
                            <NUMBER-OF-REPETITIONS>0</NUMBER-OF-REPETITIONS>
                          </EVENT-CONTROLLED-TIMING>
                        </TRANSMISSION-MODE-TRUE-TIMING>
                      </TRANSMISSION-MODE-DECLARATION>
                    </I-PDU-TIMING>
                  </I-PDU-TIMING-SPECIFICATIONS>
                  <I-SIGNAL-TO-PDU-MAPPINGS>
                    <I-SIGNAL-TO-I-PDU-MAPPING>
                      <SHORT-NAME>FrontCamera</SHORT-NAME>
                      <I-SIGNAL-REF DEST="I-SIGNAL">FrontCamera</I-SIGNAL-REF>
                      <PACKING-BYTE-ORDER>OPAQUE</PACKING-BYTE-ORDER>
                      <START-POSITION>0</START-POSITION>
                      <TRANSFER-PROPERTY>TRIGGERED</TRANSFER-PROPERTY>
                    </I-SIGNAL-TO-I-PDU-MAPPING>
                  </I-SIGNAL-TO-PDU-MAPPINGS>
                  <UNUSED-BIT-PATTERN>255</UNUSED-BIT-PATTERN>
                </I-SIGNAL-I-PDU>
```
The file is not from GM and seems to be Autosar 4.1.1 with ethernet only communication. So I guess it makes sense that there are no can ids and also the conversion to dbc makes little sense.

closed 2016-01-12 by jmtatsch