ebroecker/canmatrix #27: Quoted strings when importing

altendky, 2016-01-16:
A coworker happened across canmatrix the other day and asked if I'd heard of it...  :]  and he also had a `.sym` file that broke the importer.  Basically, there doesn't seem to be any awareness of quoted strings in the few places this exposed trouble.  Commas in enumeration items and spaces in variable names.  I asked real quick over on `#python` and the recommendation for a lightweight (and correspondingly not-so-robust) solution was `tea.shutil.split()`.

```
>>> mystring = 'a="1", b="2, 0", c="3"'
>>> mystring.split(',')
['a="1"', ' b="2', ' 0"', ' c="3"']
>>> [s.rstrip(', ') for s in tea.shutil.split(mystring)]
['a=1', 'b=2, 0', 'c=3']
>>> [(s[0], s[1]) for s in [s.rstrip(', ').split('=') for s in tea.shutil.split(mystring)]]
[('a', '1'), ('b', '2, 0'), ('c', '3')]
```

Has this already been dealt with elsewhere?  Any opinions on how locally or globally you might like to have this addressed?  I assume that there are big parsing/grammar libraries but I'm not sure either of us want to get into that.

ebroecker, 2016-01-17:
Hi,

I looked a bit around.
Some people suggest using "csv" module. I also found some regexp solutions. 

I think only sym format and maybe dbf format have currently this problem. 

Solve this issue globally or just for sym? 
I'd suggest to solve it for sym first. If we find this issue later in some other format, we can think about again...

altendky, 2016-01-24:
`csv` doesn't seem to handle partially quoted elements (`1="red blue"` as opposed to `"1=red blue"`).  [`shlex.split()`](https://docs.python.org/3.5/library/shlex.html#shlex.split) does do an alright job with a little help.  A preliminary attempt is available in aad0132c2c9006d3fa3481491158a47eed3becbe.  There is some uncertainty as to whether names like "Blue Red" are supported in the `dbc` format.

Eduard, the quoted dbc is at https://github.com/altendky/canmatrix/blob/aad0132c2c9006d3fa3481491158a47eed3becbe/test/reference/from_sym/test.dbc for testing when you get a chance.  Thanks.

altendky, 2016-02-03:
Working well enough for me for now.

closed 2016-02-03 by altendky