ebroecker/canmatrix #32: canconvert dbc=>xlsx error (v0.3)

nexulm, 2016-02-03:
After pip install canmatrix within a cygwin installation I get this error message now by performing canconvert xxx.dbc xxx.xlsx;
Traceback (most recent call last):
  File "/usr/bin/canconvert", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/site-packages/canmatrix/convert.py", line 173, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
  File "/usr/lib/python2.7/site-packages/canmatrix/convert.py", line 93, in convert
    ex.exportXlsx(db, outfile, **options)
  File "/usr/lib/python2.7/site-packages/canmatrix/exportxlsx.py", line 312, in exportXlsx
    sigHash["%02d" % int(sig.getMsbStartbitReverse()) + sig._name] = sig
AttributeError: 'Signal' object has no attribute 'getMsbStartbitReverse'
Exception Exception: Exception('Exception caught in workbook destructor. Explicit close() may be required for workbook.',) in <bound method Workbook.__del__ of <xlsxwriter.workbook.Workbook object at 0xff494a8c>> ignored

canconvert xxx.dbc xxx.xls is working fine with the same dbc.
The "old" version from last year was working fine for the conversion from dbc to xlsx with the same dbc input file

ebroecker, 2016-02-03:
could you try current master?

nexulm, 2016-02-03:
I was trying the master from this morning before with this error (stil the same after git sync with  commit 3da0ab3):
Traceback (most recent call last):
  File "/cygdrive/v/CANtools/canmatrix/canmatrix/convert.py", line 26, in <module>
    from .log import setup_logger, set_log_level
ValueError: Attempted relative import in non-package

Due to be unsure what is the reason for that issue I've used the recommended installation via "pip install canconvert" this morning.

nexulm, 2016-02-03:
fixed in 3da0ab3

commit 3da0ab39 (2016-02-03): fixed typo

may fix #32
closed 2016-02-03 by nexulm