ebroecker/canmatrix #35: Question: What do you guys use for encoding decoding can data frames?

reinzor, 2016-02-12:
Hi all,

I was wondering what you guys use for encoding/decoding the 8 bytes into signals and visa versa? I have seen this library: https://github.com/openxc/bitfield-c but I cannot get it to work properly for all different DBC formats (little/big endian, signed unsigned etc).

For python, I use https://pypi.python.org/pypi/bitarray/0.8.1 which works fine. However, my implementation is too slow if I have to handle 2000 frames * x signals per second.

I'm interested in what kind of libraries you guys use, C, C++?

Thanks!

altendky, 2016-02-13:
My application is [available on GitHub](https://github.com/altendky/st/tree/b5676012e9c2d8a4d09610f1ea515dd61e968d74) as well.  I am using a modified branch of `bitstruct` (referenced in my [`venv.py`](https://github.com/altendky/st/blob/b5676012e9c2d8a4d09610f1ea515dd61e968d74/venv.py#L113)).  Be sure to look at the [endian branch](https://github.com/altendky/bitstruct/tree/b0b13785630dc10e749f89f035deb2b9be18601e).  I do have performance issues but 1) I think they are related to the excessive Qt model/view interactions and 2) I have only about 100 frames per second.

My code needs quite a bit of organizing, tidying, and isolation from the canmatrix objects but the [canneo](https://github.com/altendky/st/blob/b5676012e9c2d8a4d09610f1ea515dd61e968d74/epyq/canneo.py) file is the heart of my [un]packing of CAN messages.

reinzor, 2016-02-13:
Thanks for sharing! However, I'm still looking for a neat c, cpp
implementation. Does anybody have any pointers for this?

-Rein

altendky, 2016-02-13:
I am not familiar with any in particular.  Are you sure optimizations of Python libraries would not be sufficient?  Or is this for a C/C++ application...  If you don't find a dedication library you may find it useful to snag code out of [CanFestival](http://www.canfestival.org/) or [CANopenNode](https://sourceforge.net/projects/canopennode/) (both are C CANopen stacks).

reinzor, 2016-02-13:
Thanks, I don't think that a python implementation would be sufficient.
And, I would like to implement the application in C++. I'm interested to in
automatically decoding can signals based on a vector can dbc converted to a
json file created with use of the conconvert tool of canmatrix. The
application should work with arbitrary DBC files of vendors.

So basically:

./app [can_interface] [path_to_dbc]

std::cout << Decoded CAN Frames std::endl;

My current python implementation is able to do this but is too slow when we
are talking about 2k frames per second.

-Rein

reinzor, 2016-02-13:
Another ref:

I would like to create a CAN_ROS_BRIDGE such as:

https://github.com/ros-industrial/ros_canopen/tree/indigo-devel/socketcan_bridge

But instead of publishing raw CAN Frames over a topic, I would like to
publish decoded frames.

-Rein

ebroecker, 2016-02-13:
I developed once some c-code for signal-decoding but never tested it. 
If you are interested in the snippets, I could provide it ...

Ben15550, 2016-02-13:
Send it.

ebroecker, 2016-02-13:
https://gist.github.com/ebroecker/2e555829eaf7454a0f4d
not testet - so may need some attention

reinzor, 2016-02-15:
I forked your gist, made it compile and added tests with expected output:

https://gist.github.com/reinzor/1ed4243785d1a27906fc

Does not seem to work properly  yet ...

-Rein

reinzor, 2016-02-15:
I am not really into the bit shifting operators etc. so if you guys could
help get this to work, it would be great!

-Rein

ebroecker, 2016-02-17:
I forked your code again and did some corrections.
currently only one error is left (one should check if its an error, or the test is wrong ...)

reinzor, 2016-02-17:
I forked your changes again and did some correction on the end bit of the
big endian. Also made the output/mask 64bit (max size can frame):

All tests seem to pass:

https://gist.github.com/reinzor/6659f260c9e9a5db8fe5

g++ signaldecode.cpp -o signaldecode
./signaldecode

Now the only thing that is left is the encoding ;).

-Rein

ebroecker, 2016-02-18:
encoding is done in my gist now

reinzor, 2016-02-18:
Thanks, I will check out tomorrow.

-Rein

reinzor, 2016-02-26:
Updated the decoding function, the extractSignal function did not take the
sign into account.

https://gist.github.com/reinzor/6659f260c9e9a5db8fe5

When I have more time, I will take a look at your encoding.

-Rein

reinzor, 2016-02-26:
@ebroecker

I created a new repo for this and added you as collaborator:

https://github.com/reinzor/libcan-encode-decode

More easy way to manage the code

-Rein

ebroecker, 2016-02-28:
@reinzor 

Nice work.

I used your library to get some old code work again:
https://github.com/ebroecker/SocketCandecodeSignals

This is a little program to decode signal values in combination with "candump" from socketcan tools.
For testing this program I used your test-frames and test-signals. Thereby I created a dbc-file (ccl_test.dbc) which represents your tests. Maybe you have use for it.

Eduard

reinzor, 2016-03-02:
Nice, thanks. If I have time coming weeks I will update the library also
for encoding and add tests for this.

-Rein

laercio-barbosa, 2016-05-24:
Hello guys

Googling for a Vector decoder/encoder in C/C++ I reached this issue 35. I get a snippet from 
[libcan-encode-decode/include/can_encode_decode_inl.h](https://github.com/reinzor/libcan-encode-decode/blob/master/include/can_encode_decode_inl.h)

To be more specific, the 'storeSignal' function. There is a error there. The line
`frame[start_byte] |= value >> startbit_in_byte;`

must be `frame[start_byte] |= value << startbit_in_byte;`

In a few days I will implement the sign extension that is marked as TODO. Then I will share with you so that you can update the library.

Thanks for this nice job!

Laercio

ebroecker, 2016-05-24:
Hi Laercio,

Thanks for the hint!
Till now no one uses the storeSignal function.

looking forward on your improvements!

Eduard

reinzor, 2016-05-25:
Hi Laercio,

Same holds for me. I've only been using the extractSignal up untill now.
Thanks already for taking the effort!

-Rein

ebroecker, 2016-06-01:
I close this issue, because it is solved here:
https://github.com/reinzor/libcan-encode-decode

closed 2016-06-01 by ebroecker