ebroecker/canmatrix #37: Missing byteorder conversion for multiplexed signals

koa43556, 2016-03-30:
Hello together.
As I am working with canmatrix from time to time I figured out a DBC file importing problem. I would like to share my finding:

The conversion of the MSB-startbit while DBC-import is not done for multiplexed signals.
`tempSig.setMsbStartbit(int(temp.group(2)))` from importdbc.py:113 is missing for multiplexed signals a few lines below.

Thank you for providing and maintaining this great collection of python code which really enlightens the work with CAN databases.

Greetings
Andi

ebroecker, 2016-03-30:
Hi Andi,
I think this is fixed already in the branch "api-cleanup".
I'll doublecheck after merging this branch to master.

ebroecker, 2016-04-05:
should be fixed with 237a9d6

closed 2016-04-05 by ebroecker