ebroecker/canmatrix #38: Question: Converting from Vector .dbc to .json I only get a subset of fields.

milhead, 2016-03-30:
I got the tool installed and working running well but it's only converting a subset of the fields in my .dbc file.  I wanted 'Comments', 'PGN', and 'priority' among others that are missing..  Am I missing some options or does it only mine specific fields

The 4 images show all the interesting fields in the .dbc file then I show the output of the .json

From the CANdb++ edit I get data...
[image]

[image]

[image]

[image]

All that I get out of the converter is:
`     {
            "id": "0x18d0fefe",
            "name": "CabIlluminationMessage",
            "signals": {
                "0": {
                    "bit_length": 8,
                    "factor": "0.4",
                    "name": "IlluminationBrightnessPercent",
                    "offset": "0"
                },
                "8": {
                    "bit_length": 8,
                    "factor": "0.4",
                    "name": "SwitchBacklightIlluminPercent",
                    "offset": "0"
                }
            }
        }
`

ebroecker, 2016-03-30:
Currently json support is very basic, if you need more fields in the json, I can add these fields quite easy. I'll have a look next weak!

ebroecker, 2016-04-08:
Hi,

I added a new option: "--jsonExportAll" 
This currently enables the export of the signal-comments, frame-comments signal attributes.
Maybe this helps you already.

currently these attributes are not imported, only exported.
Next will be to export more details. The goal should be a nearly complete export.

milhead, 2016-04-08:
Hi Eduard,

I'll try to give it a try this afternoon!

Currently I am only exporting anyway.

Thanks!

Mil

ebroecker, 2016-04-24:
was the former patch helpful or do you need more attributes?

milhead, 2016-04-24:
I finally got off my gauge updates yesterday and will give a whirl tomorrow!

Stay tuned and thanks!

Miller

milhead, 2016-04-26:
Greetings Eduard!

The new patch appears to correctly generate the json for all the fields I
am using at the current time! Excellent work and thank you for the update.

I now have work to do and can toss all the manual stuff I had added in the
past, I had set this project down for a bit but will be diving back in soon.

Miller

ebroecker, 2016-04-26:
Thanks for the feedback!

closed 2016-06-01 by ebroecker