ebroecker/canmatrix #39: Use data type and factor to calculate min/max instead of just 0/1

altendky, 2016-04-10:
https://github.com/ebroecker/canmatrix/blob/d69d7615e2e2758e66e9c5e9784235f5e45ef436/canmatrix/importsym.py#L142

Instead of just defaulting min and max to 0 and 1 for signals what do you think of setting them based on the number of bits, signedness, and factor for a signal?

altendky, 2016-04-21:
If this sounds good to you, I'll add it.  Just thought I would check first.

ebroecker, 2016-04-21:
maybe you could do this optional. 

just add it.

altendky, 2016-04-23:
I'm normally all for making things optional but i'm not sure what the use case is for preferring 0 and 1 as defaults.  Very few signals will have that range (booleans, unsigned 'percentages', etc).  Did you have something in mind where 0 and 1 would be common?

altendky, 2016-04-24:
https://gist.github.com/altendky/c5150ca098732513287fdd29dd3c9a1e

There's what I've come up with so far.  It minimizes the intrusion into each import file (only change the default value) and could be adjusted to take those defaults as parameters or have a single 'autorange' option to choose between `None` and `0`/`1`.

I know you said to just add it, but since I'm questioning the 'optional' aspect...  I'll wait for your opinion.

ebroecker, 2016-04-24:
This looks good!
I did a quick look over the other importers, and it seems this won't corrupt any of them.
Next steps could be to add the new behaviour to all importers and at last it would be nice to have something like "--recalcMinMax" in convert.py.

altendky, 2016-04-24:
I think we may not have the same understanding of my intended functionality.  I'm not looking to 'recalculate' the limits if they are specified in the file.  If the file has a limit then it will be retained.  I only want to fill unspecified limits with the natural bounds defined by the signal length, signedness, scaling, and offset.

To avoid overwriting a limit read from the file I only calculate the limit if min/max are still `None` when done parsing the signal.  If there were to be an option I would think that it would be to leave `None` and not calculate, but not to leave `0` or `1` which offer the user no opportunity to understand if a limit was found in the file or not.

If this doesn't seem clear, perhaps a hangouts chat would help us get on the same page.

ebroecker, 2016-04-25:
Don't worry, we do have the same understanding :-)

I only wanted to make the feature usable for corrupt/wrong valued databases.
But anyway, this is not important.

altendky, 2016-04-26:
I think I get it now and would propose these modes with the first being the default (though I would use the second for my application).

* Nothing (but leave `None` rather than `0`/`1`)
* Calculate limits unspecified by the file
* Force recalculation of all limits regardless of specification in file

The one thing I don't find useful to retain is the `0`/`1` option as it is now (sorry :] ).  Mostly it just masks the application from knowing what was specified and what was not.

The `Signal` member functions would remain available for use by the application for any more complicated scenarios a user may want to implement.

Does this sound like it would cover your suggestion as well as my case?  Thanks for the discussion.

ebroecker, 2016-04-26:
You are right, the `0/1`  defaults are useless. 
Lets just remove them.

If the input file does not have any min/max values specified, we should calculate them.

And at last - this is not high priority - I'll create an command line option to force recalculation (just for buggy files)

closed 2016-06-01 by ebroecker