ebroecker/canmatrix #40: Notify of invalid named parameters

altendky, 2016-04-23:
As I started into issue 39 I first updated the canmatrix I was using (normally stick with a single revision until I need to change something) and ran into some changes I needed to make to my application.  In making the changes I mistyped a few of the named parameters that are now handled via `**kwargs`.  I didn't get any notification like a normally named argument would provide and ended up down various wrong paths before I found my mistake.  So, back to the title, it would be nice to throw an exception on invalid function parameters.  As I did this example I also converted it to a much more compact and automated form that may or may not interest you.

https://gist.github.com/altendky/b2e4ad619ec738b462d49f1b1082201b

If you like this, I might convert the various places you have the new `**kwargs` form, or I might not get to it for a bit.  Time will tell.  :]

It's a bit less legible for the `multiplex` case but since that's one out of 12 parameters or so, I think it may be worthwhile.

ebroecker, 2016-04-23:
Hi,

your solution looks nice. Just convert - if you find time for it.

I think the code which uses canmatrix-objects gets more clear with named 
arguments.
In deed I am not sure if it was the best pythonic way to do define a 
nice interface.

closed 2016-11-08 by ebroecker

altendky, 2017-07-14:
Since this I learned about [attrs](https://pypi.python.org/pypi/attrs)...  I was basically recreating a bad version of a small part of that lib.  :[  @ebroecker, you should take a look at it.  It's pretty pleasant to use.