ebroecker/canmatrix #42: Decoding signed vs unsigned inputs

mwho, 2016-05-03:
It would be very helpful to have another column in the output csv file indicating signed or unsigned for every signal. 

For example:

A) SG_ my_signal : 16|16@1- (1,0) [-600|1000] "m" Vector__XXX
B) SG_ my_signal_2 : 0|5@1+ (1,0) [0|31] "m" Vector__XXX

In A), the "-" that follows "@1" indicates that this signal is signed. Likewise in B), the "+" that follows the @1 indicates that this signal is unsigned.

ebroecker, 2016-05-04:
The csv-export lacks a lot of information from other database formats.
It is not a big deal to add the sign-info. This info is in the internal database.
Maybe I find some time to add the sign info next week.

ebroecker, 2016-05-25:
just added sign support. there is a new col "is signed". "u": unsigned; "s": signed

mwho, 2016-05-31:
Thanks. Will this be making it into the next release soon?

ebroecker, 2016-06-01:
It will be in the next release.
I can - if needed - create a new release the next days. 
Do you need it released?
It also works, if you clone the current master ...

mwho, 2016-06-01:
It would be nice to have it released so I can use "pip install canmatrix" as part of an automatic deployment.

ebroecker, 2016-06-01:
OK - I'll do a release the next days - maybe at weekend.

ebroecker, 2016-06-05:
fixed with release 0.4

closed 2016-06-05 by ebroecker