ebroecker/canmatrix #50: Hints of removing some signal attributes?

miraculixdruid, 2016-07-14:
Hi
I need to remove some signal attributes, but can not figure out how to do this.
Any hint is much appreciated :)

See below for my noobish try :)
unwanted_attributes is list of ...... unwanted attributes

def replace_unwanted_signal_attributes(frame,replace):
    unwanted_attributes = ['A','B','C']

    for signal in frame._signals:
        for attribute in signal._name, signal._attributes:
            for elem in attribute:
                if elem in unwanted_attributes:
                    attribute[elem] = replace
    return frame

altendky, 2016-07-14:
I'll provide a better response shortly, but do you want to replace the value of an attribute with a particular name, or delete the attribute entirely?

miraculixdruid, 2016-07-14:
Hi. I want to delete the attribute completely.

ebroecker, 2016-07-14:
what about:
`if elem in unwanted_attributes:
                    del attribute[elem]`
I can't test at the moment...

altendky, 2016-07-14:
```
matrix_rx = list(importany.importany(self.can_path).values())[0]

def replace_unwanted_signal_attributes(frame):
    unwanted_attributes = ['LongName']
    for signal in frame._signals:
        signal._attributes = {k: v for k, v in signal._attributes.items()
                              if k not in unwanted_attributes}

    return frame

frame = matrix_rx._fl._list[0]
print(frame._signals[0]._attributes)
replace_unwanted_signal_attributes(frame)
print(frame._signals[0]._attributes)
```

Prints:
```
{'LongName': 'Enable', 'GenSigStartValue': '0.0'}
{'GenSigStartValue': '0.0'}
```

altendky, 2016-07-14:
Eduard, that approach yielded RuntimeError: dictionary changed size during iteration” (or similar) for me.  I suspect there is a way to iterate and delete efficiently but unless it's needed I won't dig more.

ebroecker, 2016-07-14:
OK - maybe I should add an api-call for it?
signal.delAttribute(attributeName)

miraculixdruid, 2016-07-14:
I also get the same runtime error.
A delAttribute method would have been really nice :)

altendky, 2016-07-14:
miraculixdruid, did my function do what you needed?

miraculixdruid, 2016-07-14:
Not yet... thing is tham I am merging several DBCs where some of them have strange signal-attributes created by some other tool. After merge I get erros related to the attributes when opening the merged DBC in CANdb++. If I manually delete these attribs first, then merge with canmatrix it works.

Now, next thing is to understand how to replace a signal in a frame.... maybe I am doing this the wrong way around....

miraculixdruid, 2016-07-14:
BTW: canmatrix is a really fantastic tool !

ebroecker, 2016-07-14:
Do you merge with convert? If so, use as main-input a dbc with all attributes, because during merge the attributes are not 'merged' jet. If this leads to your problem, this would motivate me to get the attribute-merge work...

miraculixdruid, 2016-07-14:
Hi, no I have merged with copyFrame.... frame-by-frame, dbc-by-dbc :)

miraculixdruid, 2016-07-14:
Get the same error when merging using convert :(
Error: "Database: illegal mode, line <nnnn>: <unwanted_attribute>"

ebroecker, 2016-07-14:
Yes, you are right. 
I just looked - convert does the same as you do manual: "copyFrame.... frame-by-frame"

Could you provide an example attribute wich leads to an error?
Or maybe can you even provide an example dbc (by mail)?
Usually candb++ should be able to read all attributes - I fear that copyFrame is somehow buggy :-(

1. Anyway: I'm gonna add delAttribute methodes 
2. I'd like to add some commandline option (to convert) for deleting unwanted attributes

ebroecker, 2016-07-15:
@miraculixdruid:
please checkout new features of convert
 
`--deleteSignalAttributes` and maybe `--deleteFramelAttributes`

hope that solves your issues?

miraculixdruid, 2016-07-16:
Hello

Thank you for great support.

I am currently on summer vacation, and can not test this until i am back in business.
Sorry for not being able to test this right away :(

Br
//M

ebroecker, 2016-08-21:
please check if branch stableApi (at 29efc5c or later).
you should use `--deleteSignalAttributes` and maybe `--deleteFrameAttributes` 
plus `--deleteObsoleteDefines`. This should fullfill your need.

**please reopen issue if not working**

closed 2016-08-21 by ebroecker

miraculixdruid, 2016-08-22:
Great. I'll test that asap.