ebroecker/canmatrix #54: AutoSAR version targeted in conversions??

blaber, 2016-07-27:
I'm performing python convert.py -vv x.dbc x.arxml but the resulting file doesn't contain a schema header.  So I'm trying to guess the version and paste in a header in order to load it into AE, but either the file fails schema validation or simply fails to load and errors out.

Should a schema header do the trick and, if so, what version would I use?  If not do you have other ideas what my problems would be?  Could indentation also be an issue?

ebroecker, 2016-07-28:
Hi blaber,

you are right, at least the schema is missing in the arxml target.
I guess there are more things missing.
Currently the arxml target is just untested with any AR tool. 
You are probably the first serious user for this target. 

My development was based on AR 3.2.
So the first step would indeed be to add a schema like following:

<AUTOSAR xsi:schemaLocation="http://autosar.org/3.2.1 AUTOSAR_321.xsd" xmlns="http://autosar.org/3.2.1" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">

The arxmls I came in contact with had some indentation also, so I do not belief the reason for the errors here.

ajbansal, 2016-07-28:
I created my own  autosar export from a dbc using a modified version of Eduards dbc class but for ethernet. So I can say for sure that we need the schema and indentation does not really matter, the indentation lxml creats just works fine.

ebroecker, 2016-07-28:
@blaber: please check if the branch "arxmlExport" works. I added the schema-reference.

@ajbansal: what did you have to do to export as ethernet? 
Is there a use case for converting dbc to ethernet-arxml? 
Maybe for developing gateways? 
If there are use cases I'd be interested to support such use cases. Maybe you could share your code snippet?

ajbansal, 2016-07-28:
@ebroecker : Yeah that is a possible use case, so people create the dbc's with some pre-defined attributes for ethernet, like IP address of nodes etc. And then you create an ethernet arxml from them.
I created the code as part of my current job, I am working with my company to see how much code I can share after desensitizing the code.

ebroecker, 2016-07-28:
@ajbansal : sounds good - looking forward for some small working example, this would be a unique feature for canmatrix ...

blaber, 2016-07-28:
@ajbanasal: So what version of autosar are you targeting in your export?

@ebroecker:  the 3.2.1 schema does read into AE successfully but is not able to create the network.  Here are the errors i get when doing the schema validation.  Any ideas?:

Error: The element 'CAN-CLUSTER' in namespace 'http://autosar.org/3.2.1' has invalid child element 'PHYSICAL-CHANNELS' in namespace 'http://autosar.org/3.2.1'. List of possible elements expected: 'http://autosar.org/3.2.1:NM-BUS-LOAD-REDUCTION-ACTIVE http://autosar.org/3.2.1:NM-BUS-LOAD-REDUCTION-ENABLED http://autosar.org/3.2.1:NM-IMMEDIATE-NM-CYCLE-TIME http://autosar.org/3.2.1:NM-IMMEDIATE-NM-TRANSMISSIONS http://autosar.org/3.2.1:NM-IMMEDIATE-RESTART-ENABLED http://autosar.org/3.2.1:NM-LOWER-CAN-ID http://autosar.org/3.2.1:NM-MSG-CYCLE-TIME http://autosar.org/3.2.1:NM-REMOTE-SLEEP-INDICATION-TIME http://autosar.org/3.2.1:NM-REPEAT-MESSAGE-STATE-TIME http://autosar.org/3.2.1:NM-TIMEOUT-TIME http://autosar.org/3.2.1:NM-UPPER-CAN-ID http://autosar.org/3.2.1:NM-WAIT-BUS-SLEEP-TIME'.
Error: The element 'PDU-TO-FRAME-MAPPING' in namespace 'http://autosar.org/3.2.1' has invalid child element 'PDU-REF' in namespace 'http://autosar.org/3.2.1'. List of possible elements expected: 'http://autosar.org/3.2.1:UPDATE-INDICATION-BIT-POSITION'.
Error: The element 'PDU-TO-FRAME-MAPPING' in namespace 'http://autosar.org/3.2.1' has invalid child element 'PDU-REF' in namespace 'http://autosar.org/3.2.1'. List of possible elements expected: 'http://autosar.org/3.2.1:UPDATE-INDICATION-BIT-POSITION'.
Error: The element 'PDU-TO-FRAME-MAPPING' in namespace 'http://autosar.org/3.2.1' has invalid child element 'PDU-REF' in namespace 'http://autosar.org/3.2.1'. List of possible elements expected: 'http://autosar.org/3.2.1:UPDATE-INDICATION-BIT-POSITION'.
Error: The element 'I-SIGNAL-TO-I-PDU-MAPPING' in namespace 'http://autosar.org/3.2.1' has invalid child element 'PACKING-BYTE-ORDER' in namespace 'http://autosar.org/3.2.1'. List of possible elements expected: 'http://autosar.org/3.2.1:TRANSFER-PROPERTY http://autosar.org/3.2.1:UPDATE-INDICATION-BIT-POSITION'.
Error: The element 'I-SIGNAL-TO-I-PDU-MAPPING' in namespace 'http://autosar.org/3.2.1' has invalid child element 'PACKING-BYTE-ORDER' in namespace 'http://autosar.org/3.2.1'. List of possible elements expected: 'http://autosar.org/3.2.1:TRANSFER-PROPERTY http://autosar.org/3.2.1:UPDATE-INDICATION-BIT-POSITION'.
Error: The element 'I-SIGNAL-TO-I-PDU-MAPPING' in namespace 'http://autosar.org/3.2.1' has invalid child element 'PACKING-BYTE-ORDER' in namespace 'http://autosar.org/3.2.1'. List of possible elements expected: 'http://autosar.org/3.2.1:TRANSFER-PROPERTY http://autosar.org/3.2.1:UPDATE-INDICATION-BIT-POSITION'.
Error: The element 'I-SIGNAL-TO-I-PDU-MAPPING' in namespace 'http://autosar.org/3.2.1' has invalid child element 'PACKING-BYTE-ORDER' in namespace 'http://autosar.org/3.2.1'. List of possible elements expected: 'http://autosar.org/3.2.1:TRANSFER-PROPERTY http://autosar.org/3.2.1:UPDATE-INDICATION-BIT-POSITION'.
Error: The element 'SYSTEM-SIGNAL' in namespace 'http://autosar.org/3.2.1' has invalid child element 'DATA-TYPE-REF' in namespace 'http://autosar.org/3.2.1'.
Error: The element 'SYSTEM-SIGNAL' in namespace 'http://autosar.org/3.2.1' has invalid child element 'DATA-TYPE-REF' in namespace 'http://autosar.org/3.2.1'.
Error: The element 'SYSTEM-SIGNAL' in namespace 'http://autosar.org/3.2.1' has invalid child element 'DATA-TYPE-REF' in namespace 'http://autosar.org/3.2.1'.
Error: The element 'SYSTEM-SIGNAL' in namespace 'http://autosar.org/3.2.1' has invalid child element 'DATA-TYPE-REF' in namespace 'http://autosar.org/3.2.1'.
Error: The element 'COMMUNICATION-CONNECTOR' in namespace 'http://autosar.org/3.2.1' has invalid child element 'ECU-COMM-PORT-INSTANCES' in namespace 'http://autosar.org/3.2.1'. List of possible elements expected: 'http://autosar.org/3.2.1:SHORT-NAME'.
Error: The element 'ECU-INSTANCE' in namespace 'http://autosar.org/3.2.1' has invalid child element 'ASSOCIATED-I-PDU-GROUP-REFS' in namespace 'http://autosar.org/3.2.1'. List of possible elements expected: 'http://autosar.org/3.2.1:DIAGNOSTIC-ADDRESS http://autosar.org/3.2.1:PDU-R-CONFIGURATION-ID http://autosar.org/3.2.1:RESPONSE-ADDRESSS http://autosar.org/3.2.1:SLEEP-MODE-SUPPORTED http://autosar.org/3.2.1:WAKE-UP-OVER-BUS-SUPPORTED'.

ajbansal, 2016-07-28:
@blaber: For my arxml I am using 4.2.2 schema

ebroecker, 2016-07-28:
@blaber : sorry - I needed to have a look in the schema: AUTOSAR_COMPACT.xsd (http://www.autosar.org/fileadmin/files/releases/3-2/methodology-templates/templates/auxiliary/AUTOSAR_TR_XMLSchemaSupplement.zip)

```
<!-- complex type for class CanTopology::CanCluster -->
<xsd:complexType abstract="false" mixed="false" name="CAN-CLUSTER">
.....
<xsd:element maxOccurs="1" minOccurs="0" name="PHYSICAL-CHANNELS">  
```
Thus for AR 3.2 `PHYSICAL-CHANNELS` is a valid child of `CAN-CLUSTER`.
Indeed this is not allowed in AR4.x

I'm thinking about supporting AR4 export. Till now I didn't start implementing it. 
Maybe this would also help you.

blaber, 2016-07-28:
Absolutely AR4 would help!  But I think it's a big step from 3.2...and might be a big, long effort??

blaber, 2016-07-28:
I'm confused though...the message says that P

blaber, 2016-07-28:
Oops, how did I close that??

Anyway, hmmm, the message says PHYSICAL-CHANNELS is invalid in 3.2.1, not AR4:

namespace 'http://autosar.org/3.2.1' has invalid child element 'PHYSICAL-CHANNELS'

ajbansal, 2016-07-28:
@ebroecker I would like to help you in this aspect.

ebroecker, 2016-07-29:
I am confused also. In the schema from autosar.org `PHYSICAL-CHANNELS` is a valid child of `CAN-CLUSTER` - seems you tool is wrong here.

Anyway: I started already with an optional AR4 export. It seems not to be too much effort. I expect to have a first test version after weekend. 
I definitively need someone to test - so your offer to help is very welcome!
So stay tuned.

ajbansal, 2016-07-29:
@ebroecker while I wait for permission from my company, I can help you with the export, I can help you test the export and possibly help with some bugs.

ebroecker, 2016-07-29:
first shot of Autosar4-export is in branch "arxmlExport"
Note: currently this branch only creates Autosar4-Files (with probably a lot of bugs) 
Autosar3 is still in code but deactivated for now.

As soon as AR4 gets working I'll create an option to select weather one wants AR3 or AR4

ebroecker, 2016-07-31:
finally I managed to get AR3 and AR4 passing the schema validation.

in branch "arxmlExport" only 

you can force convert to create AUTOSAR4 with the command line option "arxmlExport":
`convert.py --arxmlExportVersion 4.2 source.dbc dest.arxml`

Note: at the moment the first digit of the given AUTOSAR-Version is only used to determin wether to use AR3 or AR4 code. 
So   `--arxmlExportVersion 4.2 ` is the same like` --arxmlExportVersion 4.1.0` or `--arxmlExportVersion 4.9`.

blaber, 2016-08-01:
Thanks @ebroecker, I'll give it a try.

BTW, I just don't use Python enough...so what needs to be done to get rid
of the following error again?:
  File "convert.py", line 26, in <module>
    from .log import setup_logger, set_log_level
ValueError: Attempted relative import in non-package

ebroecker, 2016-08-04:
@blaber : did you install the package (`python setup install`)
did you have any success with the updated arxml-exporter?

ebroecker, 2016-08-21:
should be fixed

** if not please reopen issue **

closed 2016-08-21 by ebroecker