ebroecker/canmatrix #57: External access to matrix objects

christiansandberg, 2016-08-10:
I would like to use the imported CanMatrix instance programmatically but currently I need to use the private attributes prefixed with _. Is there a plan to have a fairly stable API?

Either the attributes could be renamed, get/set functions could be implemented or property functions be added.

ebroecker, 2016-08-10:
Hi Christian,

While technical it is possible to access the '_'-variables from outside it's definitively a very good idea to strive to a stable API.

I don't like the idea to rename the attributes.

I'd prefer property-functions.

@altendky : what Do you think? 
@christiansandberg : what would be your favorite?

christiansandberg, 2016-08-10:
I also think property functions is the most elegant and backwards-compatible way to do it.

Maybe also add iteration support to the list type classes (FrameList, BoardUnitListe, SignalGroup, and Frame).
```python
for frame in matrix.frames:
    print(frame.name)
    for signal in frame:
        print(signal.name)
```

altendky, 2016-08-10:
Unlike other languages you don't have to make every last setter and getter function ahead of time in Python.

```python
class MyClass():
    def __init__(self):
        self.my_property = 42
```

can be refactored without user knowledge to:

```python
class MyClass():
    def __init__(self):
        self._my_property = 42

    @property
    def my_property(self):
        # do something more here like count how many accesses if you want
        return self._my_property

    @my_property.setter
    def my_property(self, value):
        # check the new value to confirm validity or notify another class or whatever
        self._my_property = value
```

In both cases you can interact with `my_property` such as:

```python
mc = MyClass()
mc.my_property = 10
a = mc.my_property
```

So, for the members that are just `_` prefixed but do not presently need to have any extra in the accessors than just the assignment and return, just remove the `_`.  They can be refactored to have accessors later using `@property`.

All that said, I have not read up on the PEP's related to this.  [PEP8](https://www.python.org/dev/peps/pep-0008/) doesn't even mention properties.  [Google's Python style](https://google.github.io/styleguide/pyguide.html#Properties) says to "Use properties for accessing or setting data where you would normally have used simple, lightweight accessor or setter methods."

For iteration?  Maybe those classes should just inherit from `list` and add their custom functions?  This would avoid creating a bunch of pass through functions like:

```python
def __len__(self):
    return self._fl.__len__()

def __getitem__(self):
    return self._fl.__getitem__()
```

ebroecker, 2016-08-11:
started with this issue in branch "stableApi"

ebroecker, 2016-08-21:
any comments/wishes to branch `stableApi`?

all attributes have their properties now.

if no futher wishes, I'll close this issue the next days and merge `stableApi` to `master`.

christiansandberg, 2016-08-22:
To me it looks good!

closed 2016-09-26 by ebroecker