ebroecker/canmatrix #62: DBC converter is not idempotent

umlaeute, 2016-09-16:
it seems to easy to create a matrix that changes when converting it to some format and then back.

Consider the following file `1.json`:

```json
{
    "messages": [
        {
            "id": 111,
            "is_extended_frame": false,
            "name": "foo",
            "signals": [
                {
                    "bit_length": 16,
                    "factor": 1.0,
                    "is_big_endian": true,
                    "is_signed": true,
                    "name": "bar",
                    "offset": 0.0,
                    "start_bit": 3
                }
            ]
        }
    ]
}
```

Now running the following:

```bash
canconvert 1.json 2.dbc
canconvert 2.dbc 3.json
canconvert 3.json 4.dbc
canconvert 4.dbc 5.json
```

will move the startbit around:
- 1.json: `start_bit=3`
- 3.json: `start_bit=19`
- 5.json: `start_bit=35`

i would expect that the data stays the same when converting between two formats.

umlaeute, 2016-09-16:
at least these formats seem to be affected:
- dbc
- sym

ebroecker, 2016-09-17:
Hi umläute,

what version do you use?
since #48 canmatrix uses a new internal startbit representation.

If you use a newer version, your json is not valid:
startbit 3 with length 16 is not valid for a big endian signal.
for little endian there is no problem anyway.
for big endian the internal starbit you probably mean is 20.

With setting the startbit to 20 and doing your conversions again, everything seems to be ok.

Starbit is a wired thing, this is why we have a wiki-page about this:
https://github.com/ebroecker/canmatrix/wiki/signal-Byteorder

So I guess we have no bug in canmatrix, but we should add a feature to recognize wrong/impossible startbits.

umlaeute, 2016-09-17:
i'm using 0.4 (as is packaged in Debian).

about the weirdness: yes, endianness in CAN is giving me a thorough headache (i have it working for now for my specific setup; no idea about it's general applicability)...though the mathwork link in the wiki speaks to me in a language i understand (thanks for pointing me there).
anyhow, it would be fantastic if canmatrix could help in actually decoding a given message :-)

in any case: while indeed `start_bit=3` is illegal  for BE (shouldn't canmatrix raise an exception in such case?), i think the second iteration (3.json: `start_bit=19`) is valid, but giving it another conversion cycle changes it to `start_bit=35`

altendky, 2016-09-17:
https://github.com/altendky/st/blob/6a52806ca72d5f8c9392ea3ab74b1f91cefa46b5/epyq/canneo.py

It's not a well defined library nor does it use canmatrix directly (takes a matrix as a parameter and then builds it's own objects) but it does unpack and pack CAN messages.  If you are interested in running the whole application to play with it, I can certainly talk you through the setup (basically run venv.py with python3) and an introduction.

ebroecker, 2016-09-17:
OK, your Issue is fixed already with the current master.
Next release will have the fix also.

Thanks for telling me, that debian includes canmatix - I didn't even know.

You are right, canmatrix should raise an exception and recognize wrong/impossible startbits.

ebroecker, 2016-09-17:
added issue for detection of wrong startbits ( #63 )

I'll close this issue soon.

closed 2016-09-26 by ebroecker