ebroecker/canmatrix #64: Retaining BA_ and BA_DEF_ in conversions

blaber, 2016-09-22:
I like starting with your xlsx format because it's a lot cleaner with Excel than mucking directly with the dbc files in a text editor.  Unfortunately, in the conversions between dbc and xlsx, the definitions such as BA_ and BA_DEF_, etc. are lost.

Is there any way to retain that information?  Or at least would it be possible to have an option that pulls them in from a specified dbc perhaps (e.g., canconvert --defs=reference.dbc source.xlsx target.dbc)??

ebroecker, 2016-09-22:
At the moment there is no complete export to xlsx possible.
As I understand you, you are looking for a format to edit dbc-files in a nicer way.
You could do this directly with a python-script :-(
Maybe sometimes we could create a more complete xlsx-export. But at the moment, I am working on refactoring the complete code. 

Your idea to pull definitions out of another dbc-file won't work (at the moment), because the attributes are signal- and frame-specific.
You want to keep the meta data, so what do you want to edit in your dbc-file?  
(Don't get me wrong, I just want to learn more about your use-case...)

blaber, 2016-09-22:
Ok, so the use case is if you wanna define a new frame(s) in a nicely
formatted Excel spreadsheet (like you already have) and then merge it into
a master dbc file.  Even if it merges in and I have to manually add the
attributes to the new frame(s) it would still be an improvement.  The main
problem is it doesn't seem possible to do such a merge and retain the
attributes of the original master dbc in the target dbc.

altendky, 2016-09-22:
Are they DBC specific attributes?  If not, then perhaps it would be better to convert to one of the other formats with a CAN-specific editor.  PEAK PCAN Symbol Editor (.sym) is freely available.

http://www.peak-system.com/produktcd/ (browse to `Tools` > `PCAN-Symbol Editor`)

Or maybe one of the KCD tools?

https://github.com/julietkilo/kcd

I don't know, but it might be worth looking at editor options for the other formats in the mean time.  Then again, they may have other conversion issues either direction as well.

Good luck.

blaber, 2016-09-22:
No, just to simply store the corresponding PGN and/or SPN (j1939).  Thanks for the kcd suggestion...maybe it has a better structure for storing such attributes...I'll check...

ebroecker, 2016-09-23:
Did you try: 

`convert original.dbc --merge yourExcel.xlsx:frame=yourNewFrame target.dbc` 

this should take your original.dbc to your target.dbc and only add the frame `yourNewFrame` from `yourExcel.xlsx` into target.dbc

blaber, 2016-09-24:
Actually, yeah, that appears to do what I was describing in the second option!  The reason it's taken me so long to respond is because my resulting dbc file had a problem in it.  What was happening is that although the Signal Attribute (BA_DEF_ SG_) for one of the signals was of type STRING:
BA_DEF_ SG_ "signame" STRING;

 the actual string value given on the corresponding BA_ attribute definition didn't convert with quotes around it:
BA_ "signame" SG_ 1234567890 ECU StringAttribute

like there were in the original dbc file before the conversion:
BA_ "signame" SG_ 1234567890 ECU "StringAttribute"

ebroecker, 2016-09-25:
seems you found 2 bugs.

for the first one i create a new issue (#65)

I have to repoduce the second one before I know whats going wrong here.

ebroecker, 2016-09-26:
fixed in branch "stableApi"

closed 2016-09-26 by ebroecker