ebroecker/canmatrix #66: python2 support

umlaeute, 2016-09-26:
it seems that the python2 support is somewhat broken with current canmatrix.

given the file `foo.dbc`:
~~~
BO_ 262 foo: 8 Vector__XXX
 SG_ bar : 40|8@1+ (0.2,0) [0|0] "%" Vector__XXX
~~~

i can run `canconvert` under python3 just fine:
~~~
$ python /usr/bin/canconvert /tmp/foo.dbc foo3_.json
$
~~~

but doing the same with python2 fails:
~~~
$ python /usr/bin/canconvert /tmp/foo.dbc foo3_.json
[...]
Traceback (most recent call last):
  File "/usr/bin/canconvert", line 11, in <module>
    load_entry_point('canmatrix==0.4', 'console_scripts', 'canconvert')()
  File "/usr/lib/python2.7/dist-packages/canmatrix/convert.py", line 220, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
  File "/usr/lib/python2.7/dist-packages/canmatrix/convert.py", line 40, in convert
    dbs = im.importany(infile, **options)
  File "/usr/lib/python2.7/dist-packages/canmatrix/importany.py", line 37, in importany
    dbs[key] = im.importDbc(infile, **options)
  File "/usr/lib/python2.7/dist-packages/canmatrix/importdbc.py", line 113, in importDbc
    receiver = list(map(str.strip, temp.group(11).split(',')))
TypeError: descriptor 'strip' requires a 'str' object but received a 'unicode'
$
~~~

should i spend time to fix this or is py2 officially unsupported (which is ok for me, i just want to know)?

altendky, 2016-09-26:
I presume we are continuing to try to support py2.  I certainly put in a good bit of effort towards that end when I added py3 support.  :]  But, we'll see what Eduard says.

umlaeute, 2016-09-26:
so is there any reason to use `map()` rather than a simple list comprehension:

    receiver = [_.strip() for _ in temp.group(11).split(',')]

christiansandberg, 2016-09-26:
Try pip install future. I've added it as a dependency for Python 2 since it can't function without it.

altendky, 2016-09-26:
I probably would have written a list comprehension like you did.  Though, I wouldn't use `_` since you are referencing it.  Maybe `s` for substring if you want something short?

umlaeute, 2016-09-26:
@christiansandberg already did that, as this is needed to get the `from builtins import *` to work. however this seems to not help with the `str.strip`

@altendky oh i really don't care about the actual variable name; `_` is nice for code-snippets pasted somewhere without context; the actual patch might use some meaningful variable name.

ebroecker, 2016-09-26:
I try to support both py2 and py3. 

Actually my py3 installation was corrupt - I just repaired 30 minutes ago.
At the moment I have the inverse problem while fighting with the "stableApi" branch (which should better named "nextApi"). Py2 works but I have some py3-problems.

ebroecker, 2016-09-26:
by the way - your example works with py2 and py3 -- if you use "stableApi"-branch

umlaeute, 2016-09-26:
thanks. 

so i'll wait until "stableApi" has been merged into "master" and tagged as a release :-)

ebroecker, 2016-09-26:
that sounds like a good plan :-)

closed 2016-09-26 by ebroecker