ebroecker/canmatrix #68: Use Canmatrix as a library

dmpoepperlavl, 2016-11-03:
Hi, I am currently writing a Python GTK+ 3 front end for Canmatrix, as I can't find any project which does something like that already. The idea is to call a Canmatrix function with a file path and recieve a Can Matrix Object:
```python
def getCanMatrixObject(infile):
    canMatrixObject = CanMatrix()
    # do stuff
    return canMatrixObject
```
In the same way I would like to call a Canmatrix function with a Can Matrix Object and a file path for the export.
```python
def exportCanMatrixObject(canMatrixObject, outfile):
    # do export
```
Then I can use the object in the GUI to visualise the data and let the user e.g. delete signals. What would be needed to achieve this? With a little help I would be more than happy to do the changes myself and submit a patch.

Greetings, Dennis

ebroecker, 2016-11-03:
So for loading use:
`dbs = canmatrix.formats.loadp(infile, **options)`

for export:
`canmatrix.formats.dumpp(outdbs, outfileName, **options)`

what you get with loadp and have to provide to dumpp is a dictionary of canmatrixes == cancluster.
If it's only one matrix in:
`dbs =={"someNameOrEmptyString":CanMatrixObject}`

in case of multible:
`dbs =={"busName1":CanMatrixObject1, "busName2":CanMatrixObject2}`

For each matrix you can for example delete signals, ecus, frames. 

you could use `canmatrix/convert.py` as reference for loading/dumping and editing

dmpoepperlavl, 2016-11-09:
Yes, this works great! Thanks for your help.

closed 2016-11-09 by dmpoepperlavl

ebroecker, 2016-11-09:
Do you plan to release your tool open source?
Let me know if so and there is a code I can test somewhere.

dmpoepperlavl, 2016-11-09:
I would love to! I'm doing this at work, so I will see if I can make it happen.