ebroecker/canmatrix #70: Error converting from arxml (to anything) using dev-0.6

nostar, 2016-12-13:
```
INFO - convert - Importing ASR.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - DEBUG 0 frames in arxml...
DEBUG - arxml - DEBUG 181 can-frame-triggering in arxml...
DEBUG - arxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - DEBUG 808 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
DEBUG - arxml - Busname: CAN_2_Cluster
DEBUG - arxml -  Speed: 500
DEBUG - arxml -  Speed: 500
DEBUG - arxml - 181 frames found in arxml
...
...
Traceback (most recent call last):
  File "/usr/bin/canconvert", line 9, in <module>
    load_entry_point('canmatrix==0.5', 'console_scripts', 'canconvert')()
  File "build/bdist.cygwin-2.4.1-i686/egg/canmatrix/convert.py", line 294, in main
  File "build/bdist.cygwin-2.4.1-i686/egg/canmatrix/convert.py", line 41, in convert
  File "build/bdist.cygwin-2.4.1-i686/egg/canmatrix/formats.py", line 54, in loadp
  File "build/bdist.cygwin-2.4.1-i686/egg/canmatrix/formats.py", line 63, in load
  File "build/bdist.cygwin-2.4.1-i686/egg/canmatrix/arxml.py", line 1558, in load
IndexError: list index out of range
```

nostar, 2016-12-13:
I have 4 arxml files.  The smallest of the four converts successfully.  This file contains 85 frames.  The other 3 that fail with the same error contain 143, 181, and 249.

ebroecker, 2016-12-14:
Hi nostar,

do you have CAN-FD frames in your ARXML? 
There is currently a restriction of max 8 bytes per frame in the code. (https://github.com/ebroecker/canmatrix/blob/35d37af702b10be5dfaaeccf7c390f0b10751b2d/canmatrix/arxml.py#L1545)

I'll fix this soon.

nostar, 2016-12-14:
these arxml files are used on a can-fd bus (higher data rate than nominal) but there should not be any frames > 8 bytes.  Is there a way to output which frame or line in the arxml is causing the issue?

ebroecker, 2016-12-14:
I just added a quickfix for this issue - hope it works

nostar, 2016-12-20:
Yes that works, thanks!

ebroecker, 2016-12-21:
ok, I close this issue

closed 2016-12-21 by ebroecker