ebroecker/canmatrix #77: Use integers for value table keys

altendky, 2017-05-16:
`Signal.multiplex` is an integer but the multiplex signal's `values` keys are strings.  These should be consistent and I recommend choosing integer (unless maybe there could be float keys?).

```diff
commit 7386a982bafdc53e1de8df7bde83101c18cc81ac
Author: Kyle Altendorf <kyle.altendorf@epcpower.com>
Date:   Tue May 16 16:28:35 2017 -0400

    Normalize value tables to have integer keys

diff --git a/canmatrix/canmatrix.py b/canmatrix/canmatrix.py
index cbff60b..84b9c7e 100644
--- a/canmatrix/canmatrix.py
+++ b/canmatrix/canmatrix.py
@@ -163,6 +163,10 @@ class BoardUnitList(object):
         return len(self._list)
 
 
+def normalizeValueTable(table):
+    return {int(k): v for k, v in table.items()}
+
+
 class Signal(object):
     """
     contains on Signal of canmatrix-object
@@ -258,7 +262,7 @@ class Signal(object):
 
     @values.setter
     def values(self, valueTable):
-        self._values = valueTable
+        self._values = normalizeValueTable(valueTable)
 
     @property
     def comment(self):
@@ -838,7 +842,7 @@ class CanMatrix(object):
         return iter(self._fl)
 
     def addValueTable(self, name, valueTable):
-        self._valueTables[name] = valueTable
+        self._valueTables[name] = normalizeValueTable(valueTable)
 
     def addAttribute(self, attribute, value):
         """
```

altendky, 2017-05-23:
@ebroecker, are you comfortable with this change?  I just wanted to give you a chance for input here before pushing.

altendky, 2017-05-23:
I should also say that I will test and check for other places to touch as well.

ebroecker, 2017-05-23:
Yes, I'm fine with it. 
Please push in 'dev-0.6'

commit 616fea45 (2017-05-23): Normalize value tables to have integer keys

Fixes #77
closed 2017-05-23 by altendky