ebroecker/canmatrix #78: Store enumeration (in addition to|instead of) signal values

altendky, 2017-05-16:
In the case of enumerations, `Signal.values` is repetitive of the enumeration definition itself.  At least it would be nice to store the enumeration name so that the name and the relationship are not lost. Maybe the values themselves could be dropped but that's not as important and I haven't considered it in detail.

```diff
commit 94e6af3c01fea949e1280b0dddfa2628d0094274
Author: Kyle Altendorf <kyle.altendorf@epcpower.com>
Date:   Tue May 16 16:29:07 2017 -0400

    Store enumeration name to signal

diff --git a/canmatrix/canmatrix.py b/canmatrix/canmatrix.py
index 84b9c7e..fab258f 100644
--- a/canmatrix/canmatrix.py
+++ b/canmatrix/canmatrix.py
@@ -203,7 +203,8 @@ class Signal(object):
             ('receiver', '_receiver', None, []),
             ('comment', '_comment', None, None),
             ('multiplex', '_multiplex', multiplex, None),
-            ('is_float', '_is_float', bool, False)
+            ('is_float', '_is_float', bool, False),
+            ('enumeration', 'enumeration', str, None),
         ]
 
         for arg_name, destination, function, default in args:
diff --git a/canmatrix/sym.py b/canmatrix/sym.py
index d43df1d..9148f83 100644
--- a/canmatrix/sym.py
+++ b/canmatrix/sym.py
@@ -407,6 +407,7 @@ def load(f, **options):
                         signal.setStartbit(startBit)
                     if valueTableName is not None:
                         signal.values = db.valueTables[valueTableName]
+                        signal.enumeration = valueTableName
   #                  signal.addComment(comment)
                     signal.addAttribute("GenSigStartValue", str(startValue))
                     frame.addSignal(signal)
```

altendky, 2017-05-23:
@ebroecker, are you comfortable with this change?  I just wanted to give you a chance for input here before pushing.

altendky, 2017-05-23:
I should also say that I will test and check for other places to touch as well.

ebroecker, 2017-05-23:
push it in 'dev-0.6'

commit dabf79e5 (2017-05-23): Store enumeration name to signal

Fixes #78
closed 2017-05-23 by altendky