ebroecker/canmatrix #81: Test script should use virtualenv/tox

altendky, 2017-05-23:
Presently the test script attempts to install some libraries to the system Python installation.  This should not be done.  Rather, it should at least use [virtualenv](https://virtualenv.pypa.io/en/stable/) and probably even [Tox](https://tox.readthedocs.io/en/latest/) for multi-python testing.  My setup isn't the best or simplest (or even used for testing) but as one possible reference I have [tox.ini](https://github.com/altendky/st/blob/acf2231cd640d0a44bb98301679cef74f233083e/tox.ini).

For clarity, note that Tox uses virtualenv for each of the test environments it creates.

ebroecker, 2018-10-11:
@altendky 

this is done meanwhile, is it?

altendky, 2018-10-11:
We have to fully switch away from the original tests, but yes.

closed 2018-10-11 by altendky