ebroecker/canmatrix #82: String attributes retain quotes after dbc.loadp()

khrvati, 2017-06-08:
When a Node, Network or Message has a STRING attribute, dbc.loadp() does not strip the enclosing quotation marks. Signal attributes are unaffected.

The bug was encountered when attempting to read, modify, then write a .dbc file. The attributes in the output file are enclosed by two quotation marks on each side, making it improperly formatted and not usable by other programs (e.g. Vector CANdb++ reports an error).

To replicate, extract the .dbc file from [quotBug.zip](https://github.com/ebroecker/canmatrix/files/1061710/quotBug.zip), then run:
```
import canmatrix.formats

db = canmatrix.formats.loadp('quotBug.dbc')['']
node = db.boardUnitByName('SampleNode')
message = db.frameById(0)
signal = db.frameById(0).signalByName('SampleSignal')
print(db.attributes)
print(node.attributes)
print(message.attributes)
print(signal.attributes)
```

The cause is probably the addAttribute methods of Signal, Frame, CanMatrix and BoardUnit classes being implemented differently. Only Signal.addAttribute strips quotation marks, the others do not.

ebroecker, 2017-06-12:
This Bug was partly fixed in https://github.com/ebroecker/canmatrix/commit/9fe5acb65eedfa9e2707aa9ce800cadec9c8b5a1

dev-v6 is better but not fixes db.attributes:
> db = canmatrix.formats.loadp('quotBug.dbc')['']
> node = db.boardUnitByName('SampleNode')
> message = db.frameById(0)
> signal = db.frameById(0).signalByName('SampleSignal')

> print(db.attributes)
{u'NetworkAttribute': u'"SomeAttribute"'}
> print(node.attributes)
{u'NodeAttribute': u'SomeAttribute'}
> print(message.attributes)
{u'MessageAttribute': 'SomeString'}
> print(signal.attributes)
{u'SignalAttribute': u'omeOtherStrin'}

ebroecker, 2017-06-12:
dev-0.6 branch should work as expected

closed 2017-08-23 by ebroecker