ebroecker/canmatrix #83: 安装中文方法解释及注意事项

qxc2016, 2017-06-16:
作者在github上给出的安装方法：

Install instructions

Install canmatrix with either "pip install canmatrix" or "python setup.py install" and "pip install -r requirements.txt"

This installs the canmatrix package into your python installation.

In addition to the canmatrix package there are 2 scripts installed with this package:

    canconvert:

    converts CAN database formats between all supported formats.

    Example: "canconvert someArSystemdescription.arxml file.dbc"

    cancompare:

    compares 2 CAN databases (of different or same format).

    Example: "cancompare file1.dbc file2.dbc"

If you are using a *NIX-System, these scripts should be callable from command line

If you are using a Windows system, these scripts are usually installed at the location of your python installation. For example 'C:\python3.4\Scripts' or 'C:\python2.7\Scripts'

========================================================================

解释：

Install canmatrix with either "pip install canmatrix" or "python setup.py install" and "pip install -r requirements.txt"

This installs the canmatrix package into your python installation.

可以通过在命令行里输入下面3个指令中任意1个指令进行canmatrix的安装：

1.“pip install canmatrix”

2."python setup.py install"

3."pip install -r requirements.txt"

上面的安装指令会把canmatrix安装到python所在位置；

因为我使用的是win10系统，打开命令行后，输入上面3个任意指令时，提示找不到对应命令，

解决办法是添加环境变量，添加环境变量办法：http://jingyan.baidu.com/article/bea41d436879a4b4c51be6f9.html

将自己电脑上python的目录添加到环境变量，

然后打开win10的命令行，进入canmatrix目录里，

输入上面3个指令的任意一个，然后将会执行canmatrix的安装，

安装完成后，会在python\scripts\目录下生成canconvert和cancompare 2个程序文件，

然后在命令行里，输入命令‘canconvert  source.dbc target.xlsx‘，即可完成转换工作，（前提是python的目录已经添加到环境变量里，且在输入命令时的目录里，有source.dbc这个文件，否则将提示找不到source.dbc文件）。

注：我自己上面3种安装方式都尝试了一下，只有方式2安装后，才能正确使用canconvert实现转换，其它安装方式，使用canconvert时，会报错，不知道为什么。

==============================================================

In addition to the canmatrix package there are 2 scripts installed with this package:

    canconvert:

    converts CAN database formats between all supported formats.

    Example: "canconvert someArSystemdescription.arxml file.dbc"

    cancompare:

    compares 2 CAN databases (of different or same format).

    Example: "cancompare file1.dbc file2.dbc"

除了安装canmatrix外，还会生成2个scripts文件，

1个是canconvert，功能是把 .dbc文件转换成支持的所有格式文件，

举例："canconvert someArSystemdescription.arxml file.dbc"

第2个是cancompare，功能是比较相同格式的两个不同文件之间的不同，

举例：Example: "cancompare file1.dbc file2.dbc"

========================================================================

If you are using a *NIX-System, these scripts should be callable from command line

If you are using a Windows system, these scripts are usually installed at the location of your python installation. For example 'C:\python3.4\Scripts' or 'C:\python2.7\Scripts'

如果你使用的是类unix系统，上面的2个scripts就可以在命令行里使用了，

如果你使用的是windows系统，这些scripts文件一般会安装到你电脑里python所有的目录，

例如：‘C:\Python.3.4\scripts’，或者‘C:\Python.2.7\scripts’

ebroecker, 2017-06-22:
I guess this is a chinese translation?
Could you provide it as patch/pullrequest for README.md?
Or perhaps as additional file for this Repo?
I'll be welcome to integrate your translation.

weiyicheng, 2017-07-30:
yes, it's the chinese translation. Good translation. Like it.
> 解决办法是添加环境变量，添加环境变量办法：http://jingyan.baidu.com/article/bea41d436879a4b4c51be6f9.html

This is the method, how to add python to your system environment.

ebroecker, 2017-11-04:
I close this issue, because I added this info to the wiki and linked it in the readme.

closed 2017-11-04 by ebroecker