ebroecker/canmatrix #85: arxml4.2 support!?

granquet, 2017-07-13:
Hello :)

I ve stumbled upon your library and I like it a lot so far ;)
Only problem I had, is that I ve been provided with ARXML files that have been generated against autosar 4.2... which doesn't seem to be supported in canmatrix (yet).

Im pretty new to the automotive world and the python world in general... but I managed to hack something to parse my files, which I commited here: https://github.com/granquet/canmatrix/commit/205f45296b26f456e641ff18993ef48bd6feb5c9

its far from anything mergeable, but I think it could be a good starting point to discuss arxml4.2 support?
I suceeded in converting multiple ARXMLs I had into dbc... all of them seem coherent except for the starting values when the byte order is Motorola (Im getting an exception with a negative shift right...).

well, feel free to comment the commit or email me directly.

Thx,
Guillaume.

ebroecker, 2017-07-13:
Hi Guillaume,

thanx for your work.
I have no AR4.2 files yet - so every Input is welcome.

I tried to add your changes in a compatible way to dev-0.6 branch.
by the way - the dev-0.6 branch is the branch for the next planned release. 
so - if possible - develope against this branch

granquet, 2017-07-19:
Im doing my best to find AR4.2 files I could share with you... but things gets easily in the way...
stay tuned :)

ebroecker, 2017-07-20:
Someone sent this to me - it should be test/test.dbc as arxml4.2
[test.arxml.txt](https://github.com/ebroecker/canmatrix/files/1161365/test.arxml.txt)

ebroecker, 2018-10-29:
pleas create new issue if still problems

closed 2018-10-29 by ebroecker