ebroecker/canmatrix #90: Formats error

vfpape, 2017-08-02:
Dear all, Hello,
Am not familiar with Python installation/execution but trying to use your tools. 
When I launch canconvert I have some error messagees, could you help me to find what is wrong in what I did ? 

Thank you a lot,

test.dbc is an existing file and test.arxml is not existing. 

C:\Users\vfpape\Desktop\canmatrix-0.5>pip install canmatrix
Requirement already satisfied: canmatrix in d:\python install\lib\site-packages\
canmatrix-0.5-py3.6.egg

C:\Users\vfpape\Desktop\canmatrix-0.5>canconvert C:\Users\vfpape\Desktop\canmatr
ix-0.5\AlexTest\test.dbc C:\Users\vfpape\Desktop\canmatrix-0.5\AlexTest\test.arx
ml
ERROR - formats - Error importing canmatrix.arxml
ERROR - formats - Error importing canmatrix.kcd
ERROR - formats - Error importing canmatrix.fibex
ERROR - formats - Error importing canmatrix.xls
ERROR - formats - Error importing canmatrix.xlsx
ERROR - formats - Error importing canmatrix.yaml

Have a nice day,
Regards,
Alex

altendky, 2017-08-02:
The short explanation is that the arxml format requires lxml as well.  The other messages like that for formats you don't need can generally be ignored.

For installation I always use [virtualenv](https://virtualenv.pypa.io/en/stable/) to create an isolated environment for each project (or copy of a project).  I personally do not activate them but just call the python from the virtualenv directly (`venv\Scripts\python` for example).

I would recommend you try the latest from the `dev-0.6` branch.  There has been a lot of development in there since 0.5 including specifically some arxml recently.  `pip` can install directly from GitHub so the equivalent of what you did but for dev-0.6 (with a virtualenv in `venv`) would be `venv/bin/pip install git+https://github.com/ebroecker/canmatrix@dev-0.6`.  But there's an extra piece you can specify to also get the option extras for arxml support, `[arxml]`.  We have to get a little special to do both the git and extras and we end up with:

```
venv/bin/pip install git+https://github.com/ebroecker/canmatrix@dev-0.6#egg=canmatrix[arxml]
```

ebroecker, 2017-11-03:
this can be closed?! If not - please reopen!

closed 2017-11-03 by ebroecker

ebroecker, 2023-01-21:
https://github.com/ebroecker/canmatrix/pull/682