ebroecker/canmatrix #91: support for multiplex in conversion to JSON

s7ygian, 2017-08-09:
Hi,

unfortunately there is no multiplex variable on signals after converting DBC to JSON files.

Would it be possible to support this?

Thanks in advance.

altendky, 2017-08-09:
Seems likely this should get support.  I did a quick check and the [DBC for the tests](/ebroecker/canmatrix/blob/dev-0.6/test/test.dbc) doesn't seem to have any multiplexed signals so we probably need to start with fixing that.

But, a lot of work has happened on dev-0.6 since the release of 0.5.  What were you running?  How did you install it?  If not dev-0.6 then see my instructions over in #90 to get it installed and let us know if it's any different.

s7ygian, 2017-08-09:
Thank you for support.

Is there an option for showing the current version number of `canconvert`?

I think I run the newest dev-0.6. I installed it via the following docker file 1 month ago:

```
FROM debian:jessie-slim

ENV CANMATRIX_WORK_DIR=/opt/canmatrix/

RUN mkdir -p $CANMATRIX_WORK_DIR \
    && cd $CANMATRIX_WORK_DIR \
    && apt-get update \
    && apt-get install unzip git zlib1g-dev libxml2-dev libxslt1-dev python-dev python-pip -y \
    && pip install lxml \
    && git clone --recursive https://github.com/ebroecker/canmatrix.git \
    && cd canmatrix \
    && python setup.py install \
    && pip install -r requirements.txt

WORKDIR $CANMATRIX_WORK_DIR
```

Within the container I call the following command:
```
canconvert --jsonExportAll /opt/canmatrix/test.dbc /opt/canmatrix/test.json
```

altendky, 2017-08-09:
You didn't checkout the dev-0.6 branch so you have master instead.  See my comments in #90 for how to install dev-0.6.

s7ygian, 2017-08-10:
I corrected the appropriate line in the docker file as follows:

```
[...]
    && git clone --recursive https://github.com/ebroecker/canmatrix.git --branch dev-0.6 \
[...]
```

Same conversion command which works in 0.5 does not longer work in dev-0.6. Now I get the following error messages:

```
ERROR - dbc - Error with Line: [u'', u'ACTIVE', u' 0 ', u'NOT_ACTIVE', u' ']
ERROR - dbc - Error with Line: [u'', u'ACTIVE', u' 0 ', u'NOT_ACTIVE', u' ']
ERROR - dbc - Error with Line: [u'', u'ACTIVE', u' 0 ', u'NOT_ACTIVE', u' ']
ERROR - dbc - Error with Line: [u'', u'ACTIVE', u' 0 ', u'NOT_ACTIVE', u' ']
ERROR - dbc - Error with Line: [u'', u'INACTIVE', u' 0 ', u'ACTIVE', u' ']
ERROR - dbc - Error with Line: [u'', u'ACTIVE', u' 0 ', u'NOT_ACTIVE', u' ']
ERROR - dbc - Error with Line: [u'', u'ACTIVE', u' 0 ', u'NOT_ACTIVE', u' ']
ERROR - dbc - Error with Line: [u'', u'INACTIVE', u' 0 ', u'ACTIVE', u' ']
ERROR - dbc - Error with Line: [u'', u'ACTIVE', u' 0 ', u'NOT_ACTIVE', u' ']
ERROR - dbc - Error with Line: [u'', u'ACTIVE', u' 0 ', u'NOT_ACTIVE', u' ']
ERROR - dbc - Error with Line: [u'', u'INACTIVE', u' 0 ', u'ACTIVE', u' ']
```

ebroecker, 2017-08-10:
there is no multiplex-support in json jet. Thus is why I marked this issue as 'Bug'.

I can't reproduce your Error:
`ERROR - dbc - Error with Line: [u'', u'ACTIVE', u' 0 ', u'NOT_ACTIVE', u' ']`

did you get it with the 'test.dbc'?

s7ygian, 2017-08-10:
Ok. No, the error occurs only in my own test.dbc with multiplexers.

ebroecker, 2017-08-10:
Could you provide an example?

s7ygian, 2017-08-10:
> Would this info be enough for your needs?

I think, it should be enough. How about multiple multiplexer signals in one CAN message?

I attached a multiplexer sample [test.dbc](https://github.com/ebroecker/canmatrix/files/1215209/test.dbc.txt). With this file you can reproduce a similar error message:

```
ERROR - dbc - Error with Line: [u'', u'TRUE', u' 0 ', u'FALSE', u' ']
```

ebroecker, 2017-08-10:
thanx,

The problem with the dbc is currently "Enviroment Variables" are not supported. 
I'll fix this, by ignoring Enviroment Variables.

multiple multiplexer signals are not supported by dbc - are they?

ebroecker, 2017-08-10:
@s7ygian please try dev-0.6 branch again. 
any comments are welcome

s7ygian, 2017-08-15:
I tried the dev-0.6 branch again. Now it works for me.

Thanks a lot! :)

closed 2017-08-23 by ebroecker