ebroecker/canmatrix #93: New --signals capability similar to --frames

blaber, 2017-08-14:
Like for frames canmatrix allows you to delete (--deletesignal) and rename (--renamesignals) signals.  This is just a request if it would be possible to add the capability to copy given signals to a new dbc file.

blaber, 2017-09-15:
To understand the value, I should mention one application for this.  For example in J1939, one subset of signals/SPNs are associated with data parameters which include a data encoding definition and an associated Parameter Group (PG).  However the primary use of the SPN is in a DTC as reported in a diagnostic message.  This is a case where a signal won't be unique to the message/frame and may not even get assigned.

Anyway, thank you for your consideration and for canmatrix.

ebroecker, 2018-03-22:
as I am idiot on j1939 - could you please describe - maybe by example - what you exactly need?

blaber, 2018-04-05:
Sorry for the really bad description; it's about general "signals" and not actually specific to J1939 at all.  So like currently there are the following capabilities:
--ecus=ECUS  Copy only given ECUs (comma separated list) to target matrix
--frames=FRAMES Copy only given Framess (comma separated list) to target matrix

but the idea of this request would be to just add the signal version of them:
--signals=SIGNALS Copy only given Signals (comma separated list) to target matrix

Hopefully that makes a little more sense.

ebroecker, 2018-04-05:
Currently in canmatrix signals are always contained by frames (like on a real CAN-bus). 
Copying a signal from one frame to another would be easy.
Copying a signal from one database is feasible but to what target should this signal copied to?

In deed dbc support 'frameless' signals, but on a can bus a signal could not be sent only a frame could be sent.

Maybe I do not really understand your needs ...

commit 9b47962d (2019-02-18): add flat signals to canmatrix and copy option to copy them 

fixes #93
closed 2019-02-18 by ebroecker