ebroecker/canmatrix #94: ARXML to XLS issue

mahanteshdh, 2017-08-14:
Hi I am using tool to convert arxml to xls, got this error, I am using following arxml version:

http://autosar.org/schema/r4.0 AUTOSAR_4-1-1.xsd" xmlns="http://autosar.org/schema/r4.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"> 

Attached error screen:

Is there size limitation?

[image]
[image]

ebroecker, 2017-08-14:
You are using version 0.5? Your ARXML contains canFD-Frames right?
Please try dev-0.6-Branch.
It's a known issue of 0.5, that canfd is not supported (frames with more than 8 bytes are buggy in 0.5)

mahanteshdh, 2017-08-15:
Yes my arxml contains CAN FD frames. I installed dev-0.6 branch and i see below issue :
[image]

ebroecker, 2017-08-15:
OK, I fixed the error, but I guess it still doesn't do what it should do?

mahanteshdh, 2017-08-15:
I adopted your changes. Thank you! 
Now i am able to generate .dbc and .xls from my arxml.
but Min and Max value for signals seems to be incorrect and also message periodicity/cycle time is not extracted. Attached screen shot for reference
Min and Max value for a signal in arxml is -4096 and 4095 respectively but dbc and xls show 0 to 8191.

[image]

ebroecker, 2017-08-15:
the warning above "no datatype found" leads to missing min/max values.

I'm not sure how to fix this.
The problem is, that the datatype for the signal is not found.
Probably the datatype is references at another place in the arxml. 
But I have no idea where this could be.
Currently I search for the datatype in SYSTEM-SIGNAL which is reverenced by SIGNAL or I-SIGNAL.

You are not able so send me a sample (maybe email - not here)?

Best Regards

mahanteshdh, 2017-08-16:
If i open signal information in Autosar explorer tool from Vector, I can see datatype info and other info highlighted in red in attached screenshot. only cylce time/ periodicity found in PDU in other screen below : 
I do not have sample arxml, but I will try to strip out information from original and create sample.

[image]

[image]

Thank you!

ebroecker, 2017-08-16:
Maybe you could look in the arxml and search for a xml-tag "DATA-TYPE-REF" and what's the parent tag named.

mahanteshdh, 2017-08-16:
I don't have DATA-TYPE-REF xml-tag in my arxml.

ebroecker, 2017-08-16:
OK, You are right. 
In AR4 there is no data-type-ref -- there are "BASE-TYPE-REF",  a "COMPU-METHOD-REF" and a "DATA-CONSTR-REF".
Currently canmatrix does **not** read these infos from AR4-Files.

give me some Time to get this working. 
How important is this for your needs?
Do you have a timelimit?

mahanteshdh, 2017-08-16:
I can wait, Appreciate your help!

ebroecker, 2017-08-29:
@mahanteshdh : maybe you could test again?
could be better now...

mahanteshdh, 2017-09-27:
I tested the new update, and found same issue with Min,Max, resolution and offset. Time period is also not extracted to XLS.

mahanteshdh, 2017-09-27:
Sorry for responding late.

mahanteshdh, 2017-09-27:
I see Min and Max value issue for signed values not for unsigned value.

ebroecker, 2017-09-29:
OK, thanx for the info.
I'm quite busy these days. 
May take some time since I'll have a closer look...

ebroecker, 2017-12-13:
sign extraction is currently not supported.
in ARXML I have to determin the base datatype and map them to "signed" or "unsigned".

ebroecker, 2018-04-06:
@mahanteshdh 
Do you have issues with the current implementation.
Should I have another look on the code or can I close this issue?

ebroecker, 2018-08-16:
please reopen, if still a problem

closed 2018-08-16 by ebroecker