ebroecker/canmatrix #95: .arxml-->.dbc offset accuracy

canshancai, 2017-08-20:
Dear,

Thanks for this converter, it's very helpful for me to convert arxml to dbc.

But one issue of usage is: when I conver my arxml signal to dbc signal, the accuracy is lost.
Such as the big offset in arxml is changed to an likey big data but not so accurate.

Thanks~

[image]

ebroecker, 2017-08-20:
Thanks die Spur Comment.
I'll check it after Holiday.

But it does not too bad currently.

ebroecker, 2017-08-22:
Offset should be:
-1391569403256000/2147483,647 = 648000000

factor should be:
648000/2147483,647 = 0,301748514316

factor seems to be ok.
offset seems to be wrong.

ebroecker, 2017-08-22:
could you try once more please?
thanx

canshancai, 2017-08-23:
@ebroecker, Dear ebroecker,
It works now, thanks a lot~

commit 83e2bcc3 (2017-08-22): max fix #95

offset seems to be better now
closed 2017-08-23 by ebroecker