ebroecker/canmatrix #96: Support for AutoSar 4.3

chrisw1704, 2017-08-30:
I have an ARXML file with I-SIGNAL-I-PDU which is causing errors in the tool.  I believe this is due it being an Autosar 4.3 generated file.

Is there support planned for Autosar 4.3?

chrisw1704, 2017-08-30:
When I run the tool selecting arxml version 4.1.0 I get the following (signal names removed)

DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - DEBUG 15 frames in arxml...
DEBUG - arxml - DEBUG 15 can-frame-triggering in arxml...
DEBUG - arxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - DEBUG 299 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
DEBUG - arxml - Busname: xxx_xxxx
DEBUG - arxml - 15 frames found in arxml

DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - No Compmethod found!! - try alternate scheme.
DEBUG - arxml - No Compmethod found!! - try alternate scheme.
DEBUG - arxml - No Compmethod found!! - try alternate scheme.
DEBUG - arxml - No Compmethod found!! - try alternate scheme.
DEBUG - arxml - No Compmethod found!! - try alternate scheme.
DEBUG - arxml - No Compmethod found!! - try alternate scheme.
DEBUG - arxml - No Compmethod found!! - try alternate scheme.
DEBUG - arxml - No Compmethod found!! - try alternate scheme.
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
DEBUG - arxml - no isignal for xxx
INFO - convert - done

ebroecker, 2017-08-30:
Did you use master or dev-0.6 branch?
If you used master please try again with dev-0.6.

your second post (AR 4.1) is ok? Because the log seems so.

we have some enhancements for ar4.2 in dev-0.6 so maybe they help you with your ar4.3-file also.

chrisw1704, 2017-08-30:
I used dev-06.  Unfortunately 4.1 is not ok as the messages produced in the DBC file are blank, I think because of the isignal missing

chrisw1704, 2017-08-30:
the CAN-FRAME triggering is ok, but the DEBUG 299 I-SIGNAL-TO-I-PDU-MAPPING in arxml ones are missing

chrisw1704, 2017-08-30:
so there are very few signals, except for 1 frame, what can I look for to find the difference for why one is ok and the others are not?

ebroecker, 2017-08-30:
did you try with --arxmlIgnoreClusterInfo ?

chrisw1704, 2017-08-30:
No difference, it seems out of all the 15 frames only 1 gets populated with messages.  Is there anything in the arxml file that may be missing?

chrisw1704, 2017-08-31:
I see you've added logging, what should i expect to see?

ebroecker, 2017-08-31:
you should see a bit more in detail whats happening.
could you provide a new log? Maybe also a log of a 4.3-input-file?

chrisw1704, 2017-08-31:
PM'd you with the log from a 4.3 input file

ebroecker, 2017-12-09:
do I have any todos here? 
Anything open? Still not working?

chrisw1704, 2017-12-11:
Hi Eduard,

Still no luck, we can't convert it, and cannot have it created with Autostar 4.2.  So unfortunately we are stuck.

Chris

ebroecker, 2017-12-11:
Do you have a up to date log? 
I added some more logging info, maybe this helps me to understand what's going wrong

chrisw1704, 2017-12-11:
Hi Eduard,

E:\scripts\arxml_to_dbc>canconvert -vv RVC_STAR_3_2020_Ecu_Extract_2017_29a0.arxml file.dbc
INFO - convert - Importing RVC_STAR_3_2020_Ecu_Extract_2017_29a0.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - DEBUG 15 frames in arxml...
DEBUG - arxml - DEBUG 15 can-frame-triggering in arxml...
DEBUG - arxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - DEBUG 299 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
DEBUG - arxml - Busname: RVC_HSVL
DEBUG - arxml - 15 frames found in arxml

DEBUG - arxml - processing Frame: RVC_RVC_HSVL_Container_ST3_1ke7mvxn5ump3r4ulo1kvxv9k
DEBUG - arxml - RVC_RVC_HSVL_Container_ST3_ContainerIPdu
DEBUG - arxml - DEBUG: AR4.x PDU RVC_RVC_HSVL_Container_ST3_ContainerIPdu no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: HU_RVC_HSVL_Container_ST3_ec9316z3zifcbedhqkpofz3x6
DEBUG - arxml - HU_RVC_HSVL_Container_ST3_ContainerIPdu
DEBUG - arxml - DEBUG: AR4.x PDU HU_RVC_HSVL_Container_ST3_ContainerIPdu no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: DIAG_RS_RVC_ExtEth_FD_ST3_729qwbhvqs4p95nfgjqerngra
DEBUG - arxml - DIAG_RS_RVC_ExtEth_FD_ST3
DEBUG - arxml - processing Frame: DIAG_RS_RVC_ExtCAN_FD_ST3_6r9tnlmi6l9ttz1zf7bhpvxfr
DEBUG - arxml - DIAG_RS_RVC_ExtCAN_FD_ST3
DEBUG - arxml - processing Frame: DIAG_RS_RVC_PTI_FD_ST3_5cgjey6gxt54wxsvs9qgt6nf0
DEBUG - arxml - DIAG_RS_RVC_PTI_FD_ST3
DEBUG - arxml - processing Frame: DIAG_RS_RVC_IntEth_FD_ST3_240rwp3fd407zam2r2ixdxh4e
DEBUG - arxml - DIAG_RS_RVC_IntEth_FD_ST3
DEBUG - arxml - processing Frame: DIAG_RQ_ExtEth_RVC_FD_ST3_ev7buktlx3r77aoda7rw8b22g
DEBUG - arxml - DIAG_RQ_ExtEth_RVC_FD_ST3
DEBUG - arxml - processing Frame: DIAG_RQ_ExtCAN_RVC_FD_ST3_57rprss7j3dcolte2b5u41kw9
DEBUG - arxml - DIAG_RQ_ExtCAN_RVC_FD_ST3
DEBUG - arxml - processing Frame: DIAG_RQ_PTI_RVC_FD_ST3_bigc66lectyw2ht7r7eapljtf
DEBUG - arxml - DIAG_RQ_PTI_RVC_FD_ST3
DEBUG - arxml - processing Frame: DIAG_RQ_IntEth_RVC_FD_ST3_9jyo2rhq2cq9jwpl361xgjr9y
DEBUG - arxml - DIAG_RQ_IntEth_RVC_FD_ST3
DEBUG - arxml - processing Frame: Meas8_RVC_ST3_17sfau7jzknv6sbcvsrcid9zf
DEBUG - arxml - Meas8_RVC_ST3
DEBUG - arxml - No Compmethod found!! - try alternate scheme 1.
DEBUG - arxml - No Compmethod found!! - fuzzy search in syssignal.
DEBUG - arxml - No Compmethod found!! - try alternate scheme 1.
DEBUG - arxml - No Compmethod found!! - fuzzy search in syssignal.
DEBUG - arxml - No Compmethod found!! - try alternate scheme 1.
DEBUG - arxml - No Compmethod found!! - fuzzy search in syssignal.
DEBUG - arxml - No Compmethod found!! - try alternate scheme 1.
DEBUG - arxml - No Compmethod found!! - fuzzy search in syssignal.
DEBUG - arxml - No Compmethod found!! - try alternate scheme 1.
DEBUG - arxml - No Compmethod found!! - fuzzy search in syssignal.
DEBUG - arxml - No Compmethod found!! - try alternate scheme 1.
DEBUG - arxml - No Compmethod found!! - fuzzy search in syssignal.
DEBUG - arxml - No Compmethod found!! - try alternate scheme 1.
DEBUG - arxml - No Compmethod found!! - fuzzy search in syssignal.
DEBUG - arxml - No Compmethod found!! - try alternate scheme 1.
DEBUG - arxml - No Compmethod found!! - fuzzy search in syssignal.
DEBUG - arxml - getSignals: found I-SIGNAL-GROUP
DEBUG - arxml - processing Frame: DIAG_RQ_GLOBAL_ExtEth_FD_ST3_i7u6k8sz6wxnjlerxd1wlngw
DEBUG - arxml - DIAG_RQ_GLOBAL_ExtEth_FD_ST3
DEBUG - arxml - processing Frame: DIAG_RQ_GLOBAL_ExtCAN_FD_ST3_5dju13d5x72skq6i2slwc37tp
DEBUG - arxml - DIAG_RQ_GLOBAL_ExtCAN_FD_ST3
DEBUG - arxml - processing Frame: DIAG_RQ_GLOBAL_PTI_FD_ST3_b0c7en6247f44iw0m5uarz48s
DEBUG - arxml - DIAG_RQ_GLOBAL_PTI_FD_ST3
DEBUG - arxml - processing Frame: DIAG_RQ_GLOBAL_IntEth_FD_ST3_a935pfe3im3bp1alod34644qz
DEBUG - arxml - DIAG_RQ_GLOBAL_IntEth_FD_ST3
DEBUG - arxml - load: no isignal for Prk_LCP_Rq_Pr5_ST3_c1d2v54jdkgdjys6ecucdq0bo_7apbwz2dcqcli74xlka39ipx7
DEBUG - arxml - load: no isignal for ROE_RVC_ST3_m29eez8qkfpa5do3hqbq1gf1_5n9jpu2pm1sjchv3m8eat6u4a
DEBUG - arxml - load: no isignal for SVS_CPF_Rq_Pr5_ST3_8fx3658n1wessgjfd6o59m4x3_4tt2o8zfkfzgyvffg20omemos
DEBUG - arxml - load: no isignal for DI_Odo_Pr5_ST3_e68xmtlkywxgxsdwekdjjhw3t_aebrlv8czt27582tqokcrlq3t
[... 2 lines cut ...]
DEBUG - arxml - load: no isignal for IS1_Stat_Pr5_ST3_3fhcjemcedo30xv7r3v74vp2y_7d1b283bzuejcru0higd4wwly
DEBUG - arxml - load: no isignal for IS1_VehAccel_Offset_Pr5_ST3_6g2f9lut1082u61cg6rqoetk7_57kxvl2lbn8pfexfjojn3mopz
DEBUG - arxml - load: no isignal for LCP_Rs_Stat_Pr5_ST3_1glxulomg02e95kz1e9igokzo_6xbj0yc150nxzmmgkmsk1bryk
DEBUG - arxml - load: no isignal for Prk_CPF_R_Stat_Pr5_ST3_702ficm01vkn45gd9p6n16oyg_90m0sxrr7f0i5fol98zg2b6hy
DEBUG - arxml - load: no isignal for PT4_PTCoor4_Pr5_ST3_af4fxb1ykieox3itqdfzs0kpb_310ab0nlx9fwu0gy8yhhv9s7y
DEBUG - arxml - load: no isignal for PT4_PTCoor_DrvPosn_Gear_Pr5_ST3_p13j4awv9g3igt1ufcpda34i_b9qfjw02uciw0vbq2jbqkq38e
DEBUG - arxml - load: no isignal for VehLvl_Stat_Pr5_ST3_d7ipvjmqd589jhk5ekrjdc0lk_77xv314m2rgsq5l2hw0u1zzc7
DEBUG - arxml - load: no isignal for TrlrHtch_State_Pr5_ST3_6wdf22pcs2lziys99sssilj9t_cmtc8a70lsrr1mvmsdzgikr8q
DEBUG - arxml - load: no isignal for VehSpd_X_Pr5_ST3_2uag8a6f7zmbnyo23yi58wjb0_7r3exp4e9oep52okddkle7dx0
DEBUG - arxml - load: no isignal for VSS_TP_Auth_ST3_6j4id7sghlvlbroli4wjozjik_73s5uq1hzaqtzg6sen7iqjdge
[... cut ...]
DEBUG - arxml - load: no isignal for VSS_TP_SharedSecret_ST3_8pj4t8t4rm3tw3pk5hj7vonti_bo21carfke4foes83343n3h4x
DEBUG - arxml - load: no isignal for VSS_TP_VIN_ST3_bz1xsv84cxzl09j028gn39p69_6esp76bueu0vngkykygf4v975
DEBUG - arxml - load: no isignal for Meas8_RVC_ST3_2ab2l92mplnefhms1pq9t7lfy_cdcuou875s9lir739u7k8gcyi
INFO - convert - done

INFO - convert - Exporting file.dbc ...
INFO - convert - RVC_HSVL
INFO - convert - 15 Frames found
INFO - convert - done

This is what I ran just now using the dev branch.

Chris

ebroecker, 2017-12-13:
Here are problems:

DEBUG - arxml - DEBUG: AR4.x PDU RVC_RVC_HSVL_Container_ST3_ContainerIPdu no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - DEBUG: AR4.x PDU HU_RVC_HSVL_Container_ST3_ContainerIPdu no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!

the `I-PDU-TAG` has no `I-SIGNAL-TO-PDU-MAPPINGS`.

I Added another guess, but this may not help.
I'm out of information, thus I have to wait till I get a AR4.3-ARXML into my hands.

chrisw1704, 2017-12-14:
Hi Eduard,

Unfortunately I don't think I can give you ours... but thanks for all your help.

Chris

ebroecker, 2018-10-29:
please open a new issue, if new information is available. 
For now I can do nothing.

closed 2018-10-29 by ebroecker