ebroecker/canmatrix #97: Formats error

TianTianlove, 2017-09-13:
Hi everyone,
    I am not very familiar with Python. The environment is right and I installed the canmatrix with the guide. But when I execute the command  "E:\dbcfile>canconvert test.dbc test2.xlsx", There is a test.dbc in the "dbcfile" folder. There are some Errors:

ERROR - formats - Error importing canmatrix.arxml
ERROR - formats - Error importing canmatrix.kcd
ERROR - formats - Error importing canmatrix.fibex
Traceback (most recent call last):
  File "D:\Program Files\Python36\Scripts\canconvert-script.py", line 11, in <module>
    load_entry_point('canmatrix==0.5', 'console_scripts', 'canconvert')()
  File "D:\Program Files\Python36\lib\site-packages\pkg_resources\__init__.py", line 570, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "D:\Program Files\Python36\lib\site-packages\pkg_resources\__init__.py", line 2687, in load_entry_point
    return ep.load()
  File "D:\Program Files\Python36\lib\site-packages\pkg_resources\__init__.py", line 2341, in load
    return self.resolve()
  File "D:\Program Files\Python36\lib\site-packages\pkg_resources\__init__.py", line 2347, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "<frozen importlib._bootstrap>", line 979, in _find_and_load
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\Program Files\Python36\lib\site-packages\canmatrix-0.5-py3.6.egg\canmatrix\convert.py", line 31, in <module>
  File "<frozen importlib._bootstrap>", line 979, in _find_and_load
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\Program Files\Python36\lib\site-packages\canmatrix-0.5-py3.6.egg\canmatrix\formats.py", line 17, in <module>
  File "D:\Program Files\Python36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 996, in _gcd_import
  File "<frozen importlib._bootstrap>", line 979, in _find_and_load
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\Program Files\Python36\lib\site-packages\canmatrix-0.5-py3.6.egg\canmatrix\xls.py", line 35, in <module>
  File "D:\Program Files\Python36\lib\site-packages\xlwt\__init__.py", line 5, in <module>
    from .Row import Row
  File "D:\Program Files\Python36\lib\site-packages\xlwt\Row.py", line 9, in <module>
    from . import ExcelFormula
  File "D:\Program Files\Python36\lib\site-packages\xlwt\ExcelFormula.py", line 8, in <module>
    from . import ExcelFormulaParser, ExcelFormulaLexer
  File "D:\Program Files\Python36\lib\site-packages\xlwt\ExcelFormulaLexer.py", line 55, in <module>
    VERBOSE+LOCALE+IGNORECASE)
  File "D:\Program Files\Python36\lib\re.py", line 223, in compile
    return _compile(pattern, flags)
  File "D:\Program Files\Python36\lib\re.py", line 292, in _compile
    p = sre_compile.compile(pattern, flags)
  File "D:\Program Files\Python36\lib\sre_compile.py", line 555, in compile
    p = sre_parse.parse(p, flags)
  File "D:\Program Files\Python36\lib\sre_parse.py", line 793, in parse
    p.pattern.flags = fix_flags(str, p.pattern.flags)
  File "D:\Program Files\Python36\lib\sre_parse.py", line 770, in fix_flags
    raise ValueError("cannot use LOCALE flag with a str pattern")
ValueError: cannot use LOCALE flag with a str pattern

I repeat to install the canmatrix, but the problem is still not fixed. Can you tell me is there any problems, and how can I solve it.

TianTianlove, 2017-09-13:
My system OS is Win10 64x

ebroecker, 2017-09-13:
Hi,
I defintively have no Idea whats going wrong here.
But maybe noone tested it with python 3.6 yet. 
Currently we try to keep canmatrix running with python 2.7 and python 3.5.

Maybe this helps
Eduard

TianTianlove, 2017-09-13:
Thanks, I changed to Python2.7 it works. ^.^

closed 2017-10-04 by ebroecker