ebroecker/canmatrix #99: Attribute Problem

MAG20, 2017-10-04:
Hello,

i have same issues found.

1. the rename of signals does not work correctly.
I use a orginal DBC-File and use your script to change the name of one signal. The converter
generate a new DBC-File, but this file is corrupt. The Asap Editor reports a error in a line and cannot open the dbc-file. The orginal File work finely.

2. I export the DBC-File to Excel File and from Excel File to DBC back. The result is: the DBC-File lost 95% of the attributes. 

Can you help in this cases. Thanks.

ebroecker, 2017-10-04:
Hi MAG20,

1. what version did you try? Did you try dev-0.6? I cannot reproduce a problem with signal renaming. Could you perhaps create an sample-dbc which leads to an error ?

2. Yes, you are right. Converting to Excel is lossy.
  Creating a complete dbc-export to excel would lead to a very large/complex excel-file.
  What Usecase do you have? Is it just to rename a Signal? Than fixing issue  1)  should help. 

Eduard

ebroecker, 2017-11-28:
@MAG20 
could you provide more info (1), otherwise I'll have to close this issue because I'm not able to reproduce...

ebroecker, 2018-01-09:
maybe 1. is fixed now?

ebroecker, 2018-03-22:
please reopen if still an issue

closed 2018-03-22 by ebroecker