ebroecker/canmatrix #100: Converting from DBC to DBC leads to syntax errors in Vector CANdb++

saleenman95, 2017-10-06:
I know that converting from a DBC to a DBC seems useless, but it was the most surefire way to narrow down the issue that any DBC file generated by 'canconvert' ends up producing syntax errors when I attempt to open the generated file in CANdb++.  

I have narrowed down the issue to an extra set of parentheses being placed on certain lines.  I can't include the whole DBC (due to proprietary information), but I can show the lines that have been changed.

  | Input DBC |   | Output DBC
-- | -- | -- | --
Line | String | Line | String
4846 | BA_ "ECU" BU_ ECM   "ECM"; | 4786 | BA_ "ECU" BU_ ECM   ""ECM"";
4857 | BA_ "ECU"   BU_ IC "DISPLAY"; | 4796 | BA_ "ECU"   BU_ IC ""DISPLAY"";
4815 | BA_   "BusType" "CAN"; | 4809 | BA_   "BusType" ""CAN"";
4818 | BA_   "DBName" "COMPANY_j1939"; | 4810 | BA_   "DBName" ""COMPANY_j1939"";
4816 | BA_   "ProtocolType" "J1939"; | 4811 | BA_   "ProtocolType" ""J1939"";

After going through and removing the extra parentheses by hand the file opens just fine and there seems to be no loss in data.  Even though the ordering of some lines seems to be switched up and even though the file is slightly smaller than the input file, all the data looks to be there.

I'm sure I could write a script to solve this, but it would be beneficial for larger databases to have this issue fixed.

Thank you!

ebroecker, 2017-10-06:
I think, this is a duplicate of #82 and fixed in dev-0.6-branch already.

created a dbc with your input to test:
[issue100.dbc.txt](https://github.com/ebroecker/canmatrix/files/1363893/issue100.dbc.txt)

which works correct with dev-0.6-branch.

saleenman95, 2017-10-09:
Ah alright.  I just tried to compile the dev-0.6-branch, but when I run canconvert, I get a bunch of errors now.  I'll admit that I am new to python and am not much of a programmer to begin with, so perhaps there is something obvious that I did wrong.

[image]

ebroecker, 2017-10-09:
was a python-3-issue.
should be fixed now.
otherwise use python2.7.

closed 2017-11-01 by saleenman95