ebroecker/canmatrix #106: Issues converting to dbf when source file has frames with DLC > 8

nostar, 2017-10-31:
Open Source version of Busmaster doesnt support DBF files with frames > 8 so when a DBF file has been generated by canconvert with any frames > 8 then Busmaster either crashes or just fails to load the DBF.  Maybe a switch can be added to force DBF frames to 8 or skip them.  The DBC to DBF converter in Busmaster skips them.

ebroecker, 2017-10-31:
add option to skip frames with dlc > 8 is easy.

nostar, 2017-10-31:
When running  canconvert --deleteZeroSignals --skipLongDlc=8 -vv file.arxml file.dbf the large message are still in the dbf file.

When running canconvert --deleteZeroSignals --cutLongFrames=8 -vv file.arxml file.dbf the messages are smaller than the original size, but most are still > 8

nostar, 2017-11-01:
Delete works now.

ebroecker, 2017-11-01:
that means "skipLongDlc" works now, but cutLongFrames doesn't work for you?

nostar, 2017-11-01:
Both options work now on an arxml file with a single cluster.  Both options crash with this output on an arxml file with multiple clusters:
  
```
Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Python27\Scripts\canconvert.exe\__main__.py", line 9, in <module>
  File "c:\python27\lib\site-packages\canmatrix\convert.py", line 322, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
  File "c:\python27\lib\site-packages\canmatrix\convert.py", line 160, in convert
    canmatrix.formats.dumpp(outdbs, outfileName, **options)
  File "c:\python27\lib\site-packages\canmatrix\formats.py", line 103, in dumpp
    dump(db, fileObject, exportType, **options)
  File "c:\python27\lib\site-packages\canmatrix\formats.py", line 79, in dump
    moduleInstance.dump(canMatrixOrCluster, fileObject, **options)
  File "c:\python27\lib\site-packages\canmatrix\dbf.py", line 362, in dump
    sign) + '{},{}'.format(float(signal.max) / float(signal.factor),
```

ebroecker, 2017-11-01:
i took your diff in the code...
thanks

closed 2017-11-08 by ebroecker