ebroecker/canmatrix #108: "Invalid literal" error when merging DBC files

saleenman95, 2017-11-01:
Using the latest 0.6 release I am unable to merge DBC files.  I have tried with both Python 3.6 and 2.7.  Below are the errors I get for both.  

**Python 3.6**

```
C:\Users\abartz\Desktop\CAN Test Grounds>canconvert --merge=test.dbc j1939.dbc target.dbc
Traceback (most recent call last):
  File "C:\Users\abartz\AppData\Local\Programs\Python\Python36-32\Scripts\canconvert-script.py", line 11, in <module>
	load_entry_point('canmatrix==0.6.1rc0', 'console_scripts', 'canconvert')()
  File "c:\users\abartz\appdata\local\programs\python\python36-32\lib\site-packages\canmatrix-0.6.1rc0-py3.6.egg\canmatrix\convert.py", line 322, in main
  File "c:\users\abartz\appdata\local\programs\python\python36-32\lib\site-packages\canmatrix-0.6.1rc0-py3.6.egg\canmatrix\convert.py", line 41, in convert
  File "c:\users\abartz\appdata\local\programs\python\python36-32\lib\site-packages\canmatrix-0.6.1rc0-py3.6.egg\canmatrix\formats.py", line 54, in loadp
  File "c:\users\abartz\appdata\local\programs\python\python36-32\lib\site-packages\canmatrix-0.6.1rc0-py3.6.egg\canmatrix\formats.py", line 66, in load
  File "c:\users\abartz\appdata\local\programs\python\python36-32\lib\site-packages\canmatrix-0.6.1rc0-py3.6.egg\canmatrix\dbc.py", line 509, in load
ValueError: invalid literal for int() with base 10: '0M'
```

**Python 2.7**

```
C:\Users\abartz\Desktop\CAN Test Grounds>canconvert --merge=test.dbc j1939.dbc target.dbc
Traceback (most recent call last):
  File "C:\Python27\Scripts\canconvert-script.py", line 11, in <module>
    load_entry_point('canmatrix==0.6.1rc0', 'console_scripts', 'canconvert')()
  File "build\bdist.win32\egg\canmatrix\convert.py", line 322, in main
  File "build\bdist.win32\egg\canmatrix\convert.py", line 41, in convert
  File "build\bdist.win32\egg\canmatrix\formats.py", line 54, in loadp
  File "build\bdist.win32\egg\canmatrix\formats.py", line 66, in load
  File "build\bdist.win32\egg\canmatrix\dbc.py", line 509, in load
  File "C:\Python27\lib\site-packages\future-0.16.0-py2.7.egg\future\types\newint.py", line 76, in __new__
    return super(newint, cls).__new__(cls, val)
ValueError: invalid literal for long() with base 10: '0M'
```

I am unsure if these errors are due to the content of the DBC files (unfortunately I only have two different files to test with, which happen to be the ones I need to merge), or a bug in the code.  

Any help would be appreciated.  Thanks!

saleenman95, 2017-11-01:
I did a bit more digging and it looks like Vector CANdb++ recognizes that these files have "symbol names that exceed the 32 character limit" which is said to cause problems with older DBC tools.  I don't have a way to fix all of them since it is literally 500+ symbol names that are too long.

ebroecker, 2017-11-01:
I'll have a Look on canmatrix regardig this issue.
Hopefully we'll find a way get around

ebroecker, 2017-11-01:
this seems to be a problem with muxed signals.
maybe there is a new/other way to define a multiplexed signal in dbc.

ebroecker, 2017-11-01:
I added debug info to developement-branch which could help us to fix your issue.
could you please try again with the version of develop branch

ebroecker, 2017-11-03:
You have a dbc with recursive multiplexed signals. 
currently canmatrix supports only one multiplexer per frame.

saleenman95, 2017-11-03:
Sorry for my late response, I just tried the command with the dev branch.  Below is the output:

```
C:\Users\abartz\Desktop\CAN Test Grounds>canconvert --merge=test.dbc j1939.dbc target.dbc
Traceback (most recent call last):
  File "C:\Python27\Scripts\canconvert-script.py", line 11, in <module>
    load_entry_point('canmatrix==0.6.1rc0', 'console_scripts', 'canconvert')()
  File "build\bdist.win32\egg\canmatrix\convert.py", line 322, in main
  File "build\bdist.win32\egg\canmatrix\convert.py", line 41, in convert
  File "build\bdist.win32\egg\canmatrix\formats.py", line 54, in loadp
  File "build\bdist.win32\egg\canmatrix\formats.py", line 66, in load
  File "build\bdist.win32\egg\canmatrix\dbc.py", line 512, in load
Exception: ('error decoding line', ' SG_ ManufacturerCodeQualifierFlag m0M : 8|1@1+ (1,0) [0|1] "" Vector__XXX\r\n')
```

Does this still seem to be a multiplexed signal issue?  

Thanks!

ebroecker, 2017-11-03:
yes - definitive: its a 'recursive multiplex'-issue:
 the 'm0M' means:  if first multiplexer (the outer one) is 0 (m0) than ManufacturerCodeQualifierFlag is a Multiplexer in this multiplexed frame (M)

This is not supported yet by canmatrix.
I'll add it some time, but its a bit more work to do

saleenman95, 2017-11-03:
Ah gotcha.  Thanks for the explanation!  If I knew how to write Python well enough I'd help, but I'd probably just make things worse... 

Thanks for looking into it!

ebroecker, 2017-11-06:
I added some (pobably buggy) support for these complexer multiplexed frames.
If you have time - you could test...

saleenman95, 2017-11-07:
I just gave it a shot and it seemed to work just fine.  CANdb++ did not throw any errors and everything looks to be there.  However the file size decreased a bit (3,348kB down to 3,294KB).  

I'll let you know if I run into any problems.

Thanks!

ebroecker, 2017-11-08:
Thanks for your feedback;

Just for Documentation of the Code:
dbf and sym seem not to support those enhanced multiplexed frames

I guess this will be a dbc-only feature

ebroecker, 2017-11-08:
there is a bug in it (writing recursive multiplexer is wrong!) - please wait till it is fixed!

ebroecker, 2017-11-08:
fixed in https://github.com/ebroecker/canmatrix/commit/7de20cb7c7988ff6636f0445d5347eed3c9b843a

please redo your merge. 
the last one created a wong dbc

saleenman95, 2017-11-09:
Alright, so I tried it again, but now it doesn't work.

I get the following errors:
"Database: unknown symbol newly defined, line 16191: ManufacturerCodeQualifierFlag"

ebroecker, 2017-11-09:
Could You diff both generated dbcs?

saleenman95, 2017-11-09:
Shoot, I overwrote the old one, and I overwrote the canmatrix source files.  So I can't run it again with the old version.  (Unless there is a way to get that)

ebroecker, 2017-11-09:
Yes it is possible:
if you cloned the git you could 
"git checkout 7e21a2c01a7b9412f3dec488295a434755b8dab2"

otherwise got to 
https://github.com/ebroecker/canmatrix/tree/cb3a7f6a3586d8c50ae597ff37debf77b8954f12

and click on "Download ZIP"

saleenman95, 2017-11-10:
Alright, so I did a diff and sent an email to you about it (since it's a lot and I didn't want to post it all here.).  Let me know what you think.

ebroecker, 2017-11-10:
Hi saleenman95,
thanks for providing theses dbcs to me.
This helps a lot.
I already found the problem in the generated dbc. 
The 'recurse' multiplexer ManufacturerCodeQualifierFlag is missing completely in the generated file.
I'll fix this till tomorrow.

ebroecker, 2017-11-10:
should be fixed now

closed 2017-11-28 by ebroecker