ebroecker/canmatrix #109: Issue merging DBF files

nostar, 2017-11-01:
Merging DBC files works, but crashes when merging DBF files:
```
canconvert --merge=file2.dbf file1.dbf newfile.dbf
Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Python27\Scripts\canconvert.exe\__main__.py", line 9, in <module>
  File "c:\python27\lib\site-packages\canmatrix\convert.py", line 322, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
  File "c:\python27\lib\site-packages\canmatrix\convert.py", line 74, in convert
    cmcp.copyFrame(frame.id, dbTempList[dbTemp], db)
  File "c:\python27\lib\site-packages\canmatrix\copy.py", line 123, in copyFrame
    attribute, sourceDb.frameDefines[attribute].definition)
KeyError: u'GenMsgSendType'
```

ebroecker, 2017-11-01:
is fixed, if your dbfs were created with canmatrix

nostar, 2017-11-02:
I just tried the latest.  My dbf files were created by canmatrix (arxml -> dbf) but it still crashes:
```
Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Python27\Scripts\canconvert.exe\__main__.py", line 9, in <module>
  File "c:\python27\lib\site-packages\canmatrix\convert.py", line 322, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
  File "c:\python27\lib\site-packages\canmatrix\convert.py", line 74, in convert
    cmcp.copyFrame(frame.id, dbTempList[dbTemp], db)
  File "c:\python27\lib\site-packages\canmatrix\copy.py", line 123, in copyFrame
    attribute, sourceDb.frameDefines[attribute].definition)
KeyError: u'GenMsgSendType'
```

ebroecker, 2017-11-02:
you have to create your dbf files again. The bug was in the dbf-module not in the compare-module...

nostar, 2017-11-02:
Oh right, ok everything works now!

ebroecker, 2017-11-02:
thanks for your feedback - I'll close this issue now.
Just reopen, if anything does not work as expected.

commit 004b854a (2017-11-01): fixes #109 if dbfs were created with canmatrix
dbf was incomplete if no default-values for defines were given
closed 2017-11-02 by ebroecker