ebroecker/canmatrix #112: ValueError cannot use LOCALE flag with a str pattern

AndreiRadac, 2017-11-16:
Hello,

I receive this error "ValueError cannot use LOCALE flag with a str pattern", when I try to run anything related to this project. For example I tried to run test.py:

 This is the output:

> :\Users\uidj5318\Desktop\canmatrix-development\test>py test.py
Traceback (most recent call last):
  File "test.py", line 10, in <module>
    import canmatrix.convert
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "C:\Program Files\lib\site-packages\canmatrix-0.6.1rc0-py3.6.egg\canmatrix\convert.py", line 32, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "C:\Program Files\lib\site-packages\canmatrix-0.6.1rc0-py3.6.egg\canmatrix\formats.py", line 21, in <module>
  File "C:\Program Files\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "C:\Program Files\lib\site-packages\canmatrix-0.6.1rc0-py3.6.egg\canmatrix\xls.py", line 35, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "C:\Program Files\lib\site-packages\xlwt_future-0.8.0-py3.6.egg\xlwt\__init__.py", line 5, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "C:\Program Files\lib\site-packages\xlwt_future-0.8.0-py3.6.egg\xlwt\Row.py", line 9, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "C:\Program Files\lib\site-packages\xlwt_future-0.8.0-py3.6.egg\xlwt\ExcelFormula.py", line 8, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "C:\Program Files\lib\site-packages\xlwt_future-0.8.0-py3.6.egg\xlwt\ExcelFormulaLexer.py", line 55, in <module>
  File "C:\Program Files\lib\re.py", line 233, in compile
    return _compile(pattern, flags)
  File "C:\Program Files\lib\re.py", line 301, in _compile
    p = sre_compile.compile(pattern, flags)
  File "C:\Program Files\lib\sre_compile.py", line 562, in compile
    p = sre_parse.parse(p, flags)
  File "C:\Program Files\lib\sre_parse.py", line 865, in parse
    p.pattern.flags = fix_flags(str, p.pattern.flags)
  File "C:\Program Files\lib\sre_parse.py", line 832, in fix_flags
    raise ValueError("cannot use LOCALE flag with a str pattern")
ValueError: cannot use LOCALE flag with a str pattern

**This is the output of : python setup.py install**

> C:\Users\uidj5318\Desktop\canmatrix-development>python setup.py install
C:\Program Files\lib\site-packages\setuptools\dist.py:331: UserWarning: Normalizing '0.6.1-pre' to '0.6.1rc0'
  normalized_version,
running install
running bdist_egg
running egg_info
writing canmatrix.egg-info\PKG-INFO
writing dependency_links to canmatrix.egg-info\dependency_links.txt
[... 92 lines cut ...]

**This is the output of: pip freeze**

> bitstruct==3.5.0
canmatrix==0.6.1rc0
et-xmlfile==1.0.1
freeze==1.0.10
future==0.16.0
jdcal==1.3
lxml==4.1.1
odfpy==1.3.5
openpyxl==2.5.0b1
pyaml==17.10.0
PyYAML==3.12
six==1.11.0
tablib==0.12.1
unicodecsv==0.14.1
xlrd==1.1.0
xlsxwriter==1.0.2
xlwt==1.3.0
xlwt-future==0.8.0

Any ideas on how to fix this issue? 

Thanks

ebroecker, 2017-11-16:
could you try an older python-version?
2.2 and 3.5 are tested.

for me it seems,  xlwt has problems.
You could also try to remove xlwt  package.

AndreiRadac, 2017-11-17:
Hello,

**On Py 3.5.3 I get this for test.py:**

> 
D:\KIT-URI\PyModules\canmatrix-development\test>python test.py
arxml -> ['csv', 'dbc', 'dbf', 'json', 'kcd', 'sym', 'xls', 'xlsx', 'xml', 'yaml']
dbc -> ['arxml', 'csv', 'dbf', 'json', 'kcd', 'sym', 'xls', 'xlsx', 'xml', 'yaml']
dbf -> ['arxml', 'csv', 'dbc', 'json', 'kcd', 'sym', 'xls', 'xlsx', 'xml', 'yaml']
json -> ['arxml', 'csv', 'dbc', 'dbf', 'kcd', 'sym', 'xls', 'xlsx', 'xml', 'yaml']
kcd -> ['arxml', 'csv', 'dbc', 'dbf', 'json', 'sym', 'xls', 'xlsx', 'xml', 'yaml']
sym -> ['arxml', 'csv', 'dbc', 'dbf', 'json', 'kcd', 'xls', 'xlsx', 'xml', 'yaml']
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
Variable type 'string' found and skipped
xls -> ['arxml', 'csv', 'dbc', 'dbf', 'json', 'kcd', 'sym', 'xlsx', 'xml', 'yaml']
xlsx -> ['arxml', 'csv', 'dbc', 'dbf', 'json', 'kcd', 'sym', 'xls', 'xml', 'yaml']
yaml -> ['arxml', 'csv', 'dbc', 'dbf', 'json', 'kcd', 'sym', 'xls', 'xlsx', 'xml']
Traceback (most recent call last):
  File "test.py", line 82, in <module>
    exit_code = subprocess.call(['diff', '-r', 'reference', 'converted'])
  File "C:\Python35\lib\subprocess.py", line 247, in call
    with Popen(*popenargs, **kwargs) as p:
  File "C:\Python35\lib\subprocess.py", line 676, in __init__
    restore_signals, start_new_session)
  File "C:\Python35\lib\subprocess.py", line 955, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified

**And this for  createTestMatrix.py:**
> 

D:\KIT-URI\PyModules\canmatrix-development\test>python createTestMatrix.py
Traceback (most recent call last):
  File "createTestMatrix.py", line 7, in <module>
    import canmatrix.exportall as ex
ImportError: No module named 'canmatrix.exportall'

pip freeze:

> 
D:\KIT-URI\PyModules\canmatrix-development\test>python -V
Python 3.5.3

D:\KIT-URI\PyModules\canmatrix-development\test>pip freeze
bitstruct==3.5.0
canmatrix==0.6.1rc0
freeze==1.0.10
future==0.16.0
lxml==4.1.1
pyaml==17.10.0
PyYAML==3.12
six==1.11.0
xlrd==1.1.0
xlsxwriter==1.0.2
xlwt==1.3.0
xlwt-future==0.8.0

ebroecker, 2017-11-17:
simply converting should work?
canconvert test.dbc test2.dbc

AndreiRadac, 2017-11-17:
Yes simply converting  is working (no error). 

I tried with _python convert.py test.dbc test2.dbc_

ebroecker, 2017-11-17:
createTestMatrix.py should work again with: 21063ed640b22c9c8c9935c257aac33e1ecadd85

AndreiRadac, 2017-11-20:
The following error is popping up: **_'str' object has no attribute 'decode'_** so I deleted the _"decode("utf-8")"_ for _unit="specialCharUnit°$", and  "Multi \n Line \n Signal comment with a-umlaut: ä")_  and now is working.

For test.py is the same behavior.

AndreiRadac, 2017-11-20:
I don't want to create another issue for this question, and I don't know where to address it so I address it here.

@ebroecker : Dose this project(package) supports CAN FD ?

ebroecker, 2017-11-27:
CAN FD is now handled in this issue: #115

ebroecker, 2017-11-28:
can this issue be closed?

closed 2017-12-09 by ebroecker

AKstudios, 2022-10-21:
I'm getting the same error on Python 3.9.2 and the latest canmatrix commit. 

> raise ValueError("cannot use LOCALE flag with a str pattern")
> ValueError: cannot use LOCALE flag with a str pattern

I'm also not able to run the test examples:
`canconvert test.dbc test2.dbc`
or
`python convert.py ..\test.dbc test2.dbc`

Both of the above return the same error.