ebroecker/canmatrix #113: Question: Why start_bit is different from the dbc file?

peppesilletti, 2017-11-23:
Hello, first of all thank you very much for the effort you all are putting in this tool, it's very helpful for converting dbc to json and make my life easier :)

I have only one question about the parsing of the dbc files. Is there any particular reason why the start_bit in the generated JSON file is different from the starting bit in the dbc file? I understand it may be related to the Endianness of the can message, but sometime it would be handy, at least for me, to have the original value and then do stuff with it. 

Thank you for your attention!

peppesilletti, 2017-11-24:
Hi again, I would like to suggest the possibility to choose if the `start bit` will be computed or provided as it is and leave the user decide what to do with it. Does it sound feasible?

ebroecker, 2017-11-24:
this is possible, we'll could add an option
I'll propose something like the --xlsMotorolaBitFormat switch for xlsx-generation

peppesilletti, 2017-11-24:
That's great, would it be also possible for the conversion from dbc to json? Something like --jsonExportAll --motorolaBitFormat

ebroecker, 2017-11-24:
that was what I ment with my prior post. I'll do this these days

peppesilletti, 2017-11-24:
oh sorry I have misunderstood, too tired today : ) That's great, I really appreciate that, thank you very much for implementing the feature : )

ebroecker, 2017-11-26:
just added option `--jsonMotorolaBitFormat`.

closed 2017-11-28 by ebroecker