ebroecker/canmatrix #115: Question: Generate CAN FD dbc file?

AndreiRadac, 2017-11-27:
How can I configure to generate CAN FD dbc files?

ebroecker, 2017-11-27:
good question.
I never dealed with CAN FD files. 
What makes them different?

AndreiRadac, 2017-11-27:
You can see some info here: https://www.kvaser.com/wp-content/uploads/2016/10/comparing-can-fd-with-classical-can.pdf
also in the image:
[image]

The main update is that CAN FD can hold more data in the CAN-frame, from 8 up to 64 byte.

ebroecker, 2017-11-27:
In Dbc this seems to be only a database attribute:
"BusType" "CAN FD"
 maybe we could collect here, what to do with dbcs

AndreiRadac, 2017-11-27:
Yes, but you need to take also in consideration that the frame is bigger, so I think that a number of changes  are needed.

Can you investigate this topic, to see what else should be changed?

ebroecker, 2017-11-28:
canmatrix-objects should have no length-restrictions.

ebroecker, 2017-11-28:
A sample-dbc with fd-frames looks like this:
canmatrix can handle this dbc 

so - currently no need to fix anything.
Is anything to do in this issue?

```
VERSION "created by canmatrix"

NS_ :

BS_:

BU_: 

BO_ 2147483651 CanExtended3: 8 Vector__XXX

BO_ 4 CanStandard4: 8 Vector__XXX

BO_ 2147483650 CanFdExtended2: 16 Vector__XXX
 SG_ signal2 : 64|64@1- (1,0) [0|0] "" Vector__XXX
 SG_ signal1 : 0|64@1- (1,0) [0|0] "" Vector__XXX

BO_ 1 canFdStandard1: 24 Vector__XXX
 SG_ signal3 : 128|64@1- (1,0) [0|0] "" Vector__XXX
 SG_ signal2 : 64|64@1- (1,0) [0|0] "" Vector__XXX
 SG_ signal1 : 0|64@1- (1,0) [0|0] "" Vector__XXX

BA_DEF_ "BusType" STRING;
```

AndreiRadac, 2017-11-29:
Do you have a CANdb++ Editor to open an CAN DBC with FD Frames ?

[image]

In "Type" you should be able to select CAN FD Standard. 

And the BusType probably should be a parameter, to generate:  **_BA_ "BusType" "CAN FD";_**

ebroecker, 2017-11-29:
you are right:
correct it should look like:

```
VERSION "created by canmatrix"

NS_ :

BS_:

BU_: 

BO_ 2147483651 CanExtended3: 8 Vector__XXX

BO_ 4 CanStandard4: 8 Vector__XXX

BO_ 2147483650 CanFdExtended2: 16 Vector__XXX
 SG_ signal2 : 64|64@1- (1,0) [0|0] "" Vector__XXX
 SG_ signal1 : 0|64@1- (1,0) [0|0] "" Vector__XXX

BO_ 1 canFdStandard1: 24 Vector__XXX
 SG_ signal3 : 128|64@1- (1,0) [0|0] "" Vector__XXX
 SG_ signal2 : 64|64@1- (1,0) [0|0] "" Vector__XXX
 SG_ signal1 : 0|64@1- (1,0) [0|0] "" Vector__XXX

BA_DEF_ "BusType" STRING;
BA_DEF_ BO_  "VFrameFormat" ENUM  "StandardCAN","ExtendedCAN","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","StandardCAN_FD","ExtendedCAN_FD";
BA_ "VFrameFormat" BO_ 2147483651 1;
BA_ "VFrameFormat" BO_ 1 14;
BA_ "VFrameFormat" BO_ 2147483650 15;
BA_ "BusType" "CAN FD";
```

AndreiRadac, 2017-12-04:
@ebroecker How did you generate this? 

Maybe you can add a Py script to generate a CAN FD template, like createTestMatrix.py.

ebroecker, 2017-12-04:
creating this, is possible with a script like createTestMatrix.py
But I'll prefere a cleaner solution: Adding an frame-attribute (member-variable) like 'is_fd' to frame-class and add support for this to dbc reader and writer.

AndreiRadac, 2017-12-04:
@ebroecker : Ok, thanks.

Short question: when do you plan to release canmatrix 0.7,in  which this solution will be added?

ebroecker, 2017-12-04:
I'll try to add this feature this week in devtree.

AndreiRadac, 2017-12-04:
Ok, I'll wait for your reply when everything is finished.

Thanks!

ebroecker, 2017-12-04:
added example: /test/createTestFdMatrix.py

AndreiRadac, 2017-12-07:
I tried to run the script and this is the output:

[image]

AndreiRadac, 2017-12-07:
I modified the script like below, instead of using **import canmatrix** and **db = canmatrix.CanMatrix()**
 
And now is working.

> import canmatrix.formats
from canmatrix.canmatrix import *
db = CanMatrix()

> myFrame1 = Frame("canFdStandard1",Id=1,  dlc=24, is_fd = True)
myFrame2 = Frame("CanFdExtended2",Id=2,  dlc=16, extended = True, is_fd = True)
myFrame3 = Frame("CanExtended3", Id=3, dlc=8, extended = True)
myFrame4 = Frame("CanStandard4", Id=4,  dlc=8)
mySignal1 = Signal("signal1", signalSize=64, startBit=0, is_little_endian=True, min=0, max=0, is_signed=True)

> mySignal2 = Signal("signal2", signalSize=64, startBit=64, is_little_endian=True, min=0, max=0, is_signed=True)
mySignal3 = Signal("signal3", signalSize=64, startBit=128, is_little_endian=True, min=0, max=0, is_signed=True)

ebroecker, 2017-12-07:
did you update the whole repo?
I modified __init__.py so that a simple "import canmatrix" should be enough.

AndreiRadac, 2017-12-07:
@ebroecker  I downloaded the whole project as a zip file, and runed again the setup.py install.

But the behavior is the same -> no attribute CanMatrix

ebroecker, 2017-12-07:
ok, thankx for your feedback. Fixed it.

can I close this issue ?

AndreiRadac, 2017-12-07:
@ebroecker Yep, now the script is working.

But if I try to convert the **testfd.dbc** in a **xlsx** file I get this.

[image]

**Do you know why?**

You can leave this issue opened, and discuss here about other changes for CAN FD.

AndreiRadac, 2017-12-07:
@ebroecker now is working, but:
- I created a DBC with createTestMatrix.py, converted to xlsx, and then try to convert from xlsx to DBC, and this error pops up:

[image]

The same behaviour is present if the DBC is a FD DBC with multiple Network nodes.

- Also in createTestMatrix.py there is:

> db.boardUnits.add(BoardUnit("testBU"))
db.boardUnits.add(BoardUnit("recBU"))

but the output in the DBC is this:

[image]

There are some network nodes out of nowhere, like **B,e,s,t,U**

AndreiRadac, 2017-12-07:
@ebroecker Now the network nodes are ok, but the problems with converting form xlsx to DBC is still present.

[image]

On your side the conversion is ok?

ebroecker, 2017-12-07:
Did you Update xlsx.py?

AndreiRadac, 2017-12-07:
@ebroecker I  runed again the setup.py install to update the egg file, and now is ok. 

**If I generate the DBC form the testfd.xlsx the "BusType" is not present in the new DBC file.  The BusType should be handled also when a conversion is done.**

ebroecker, 2017-12-09:
I added some autodetection.
If frames with dlc > 8 in xlsx, these frames will be marked as fd.
If any fd-frame in a matrix, the attributes will be generated automatically

AndreiRadac, 2017-12-11:
@ebroecker I don't think that the solution with dlc > 8 is a good one. Because there are time when in a DBC file a CAN FD frame has DLC =< 8, so that frame will not be marked as FD if the condition is DLC>8.

ebroecker, 2017-12-11:
therefor you  can use the commandline-switche `--setFrameFd=myFrame,myFrame2` which forces frames to be fd-frames.

I am currently inconclusive about an additional attribute to xlsx,xls import/export.

AndreiRadac, 2017-12-11:
@ebroecker Yes, it can be used. 

But I think that storing the frame type in an additional attribiute in the xlsx,xls is a better solution. Is a more cleaner solution, because you just configure in the xlsx,xls what frame type is and then generate the DBC (or vice versa: generate form DBC the frame type in the new attribute). 

If you use --setFrameFd everytime you need to check: myFrame is FD, so I'll added to the list, myFrame2 is not FD, so I'll not add this to the list.

It's up to you what you want to do, this is just my opinion.

ebroecker, 2018-03-22:
exporting is possible now:
canconvert --additionalFrameAttributes=is_fd some.dbc some.xlsx

Do you need import also?

ebroecker, 2018-04-06:
Add functionallity to import additional attributes also

closed 2018-07-18 by ebroecker

Inddgithub, 2022-12-04:
> BusType" "CAN FD"

Besides setting the Bus type, is it possible to set different baudrates in DBCs for arbitration and data phase, when the bit rate switch is set to 0?
This is the line for CAN 2.0:
BA_DEF_DEF_  "Baudrate" 1000000;

When this value is set in a CAN FD DBC: BA_DEF_DEF_  "Baudrate" 2000000;  
does it mean that this value refers only to the data phase, and the value for arbitration phase is the maximal available one (1 Mbps)?