ebroecker/canmatrix #118: Factor,Offset, Min,Max for a Signal

AndreiRadac, 2017-12-11:
Hello,

In the xlsx template is not specifed the Factor and Offset so how are they handled? 

And for Min-Max it is the Name / Phys. Range ? But if I use the this column for Range, how can I define a value table for the signal ? For example the range is form 0-100, but I only want to define a name for the values 1,2,3.

ebroecker, 2017-12-11:
xls(x) is a lossy format.
It is used to create a human readable _extract_ of a canmatrix.
Importing/loading a xls(x) is more a gimmik.

If you want to edit a canmatrix-object, maybe json is a better choice. 
In deed I am more open to add information to json than to xls(x).
 
is json an option for you?

AndreiRadac, 2017-12-12:
I'm just using xls(x) format, because in this format I receive the information about the DBC file.

I had a look in to the code and form what I saw, the Factor is retrieved from the column "Function/Incremental Unit": if in that column is just a number, for example: 0.1 then this will be the factor, if we have: 0.1 m^2,then 0.1 will be the factor and m^2 the unit, right? 

The offset form what I saw is the min value of the signal.

 
[image]

But for Min-Max my question is still remains the same. How can I define a value table for a signal (define names for values 1,2,3 of the signal), and also have a range.

closed 2017-12-15 by AndreiRadac

JaveCai, 2019-05-06:
Hello,

I think the "offset" value should be get from dbc file rather than using the value of "mini".

ebroecker, 2019-05-06:
@JaveCai 
this issue is about reading from xlsx.
If reading dbc offset is taken from dbc.