ebroecker/canmatrix #121: Conversion error - Critical

vrathore84, 2017-12-14:
Traceback (most recent call last):
  File "c:\program files\python36-32\lib\runpy.py", line 193, in _run_module_as_
main
    "__main__", mod_spec)
  File "c:\program files\python36-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python36-32\Scripts\canconvert.exe\__main__.py", line 9
, in <module>
  File "c:\program files\python36-32\lib\site-packages\canmatrix\convert.py", li
ne 294, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
  File "c:\program files\python36-32\lib\site-packages\canmatrix\convert.py", li
ne 41, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "c:\program files\python36-32\lib\site-packages\canmatrix\formats.py", li
ne 54, in loadp
    return load(fileObject, importType, **options)
  File "c:\program files\python36-32\lib\site-packages\canmatrix\formats.py", li
ne 66, in load
    dbs[key] = moduleInstance.load(fileObject, **options)
  File "c:\program files\python36-32\lib\site-packages\canmatrix\dbc.py", line 7
32, in load
    db.frameById(int(temp.group(2))).signalByName(
AttributeError: 'NoneType' object has no attribute 'group'

ebroecker, 2017-12-14:
you have a dbc with a strange `BA_`  line.

vrathore84, 2017-12-14:
First of all i am really glad that you have responded back so quickly.
Can you please elaborate more on ba_lines ?

ebroecker, 2017-12-14:
added an try-catch-block with debuginfo (line-number and line-content) to get more info from your dbc.

could you try the current development-branch?

vrathore84, 2017-12-15:
I replaced dbc.py in Lib\site-packages\canmatrix  from latest buils its working fine.
Thanks a lot for this support.

closed 2017-12-15 by vrathore84

ebroecker, 2017-12-15:
You should have an output?
Something like `error with line no: `

could you please have a look on it?
Because this line will be ignored, but could be used...