ebroecker/canmatrix #122: Default attributes for Messages, Signals, DB,BU

AndreiRadac, 2017-12-15:
Hello,

I found that the default attributes for Messages, Signals, DB, BU are not taken into consideration.

If an signal,message,db,bu doesn't have a specified value for an attribute the default value for the attribute should be taken in consideration.

Just an example, for frame 0x01:

> If the message has a specified value for an attribute: **BA_** "GenMsgSendType" BO_ 1 0;, the value of the attibute should be the specified one, if the message dosen't have a specified value, so the text " BA_ "GenMsgSendType" BO_ 1 0;" is not generated in the DBC, the value for that message should be taken form **BA_DEF_DEF_**  "GenMsgSendType" "Cyclic";

The same idea is for all the attributes for Messages,Signals, DB, Bu.

The simplest solution I found is to add in dbc.py a new condition in  "for define in db.frameDefines:", if the define is not in the frame attributes then add to this define the default value, and if the define is in the frame attibutes keep the value that is specifed (BA_....).

[image]

And then add a new check, in my case in the xlsx.py file, to check also for the value of the default attribute, which in this case is not a number, so for Cyclic I should check if ... == "Cyclic", not if ... ="0". Because this is what db.frameDefines[define].defaultValue[1:-1] adds to the value of the attribute.

[image]

In this example I've taken it into consideration only the type "ENUM" for the Messages attributes, but this modifications should be added to all objects: Database, BoardUnit, Messages and Signals.

@ebroecker : What do you think is the best way to do this?

ebroecker, 2017-12-16:
should be fixed for xls(x)

still open for:
* csv
* json
* kcd
* sym

AndreiRadac, 2017-12-18:
@ebroecker : I just tested this, and there is still a problem.

The value of **return define.defaultValue** is the name of the default value, for example "Cyclic". So if you do a check like **if frame.attribute(db, "GenMsgSendType") == "0":** this will not be true because the value of  frame.attribute(db, "GenMsgSendType") in case of default value is not "0", but "Cyclic". So you need to compare with **'** "Cyclick" **'**, or with "Cyclic", if you add [1:-1], to define.defaultValue to cut off **" "**.

ebroecker, 2017-12-18:
the fault is how enums are handled in canmatrix. 

This should be fixed, and comparison should be against "Cyclic" and not against numbers (which could have other values in different dbcs anyway).

AndreiRadac, 2017-12-18:
@ebroecker : Do you have an idee how to fix this fault?

ebroecker, 2017-12-18:
Yes.
I'll do

AndreiRadac, 2017-12-18:
@ebroecker : Also what I observed is that now the type of cycleTime = getIfPossible(row, "Cycle Time [ms]") is "str" so if you convert from xls(x) to DBC all the messages have "0" cycletime because of:
>  if type(cycleTime).__name__ != "float":
                cycleTime = 0.0
            newBo.addAttribute("GenMsgCycleTime", str(int(cycleTime)))

AndreiRadac, 2017-12-19:
@ebroecker : Just tested, an this is the output:

[image]

My test dbc:
> VERSION ""
NS_ : 
BS_:
BU_:
BO_ 1 New_Message_2: 1 Vector__XXX
 SG_ New_Signal_2 : 0|8@1- (1,0) [0|0] "" Vector__XXX
BO_ 0 New_Message_56: 1 Vector__XXX
 SG_ New_Signal_129 : 0|8@1- (1,0) [0|0] "" Vector__XXX
EV_ EnvKlemme30: 0 [0|1] "" 1 1 DUMMY_NODE_VECTOR0 Vector__XXX;
EV_ EnvKlemme15: 0 [0|1] "" 0 2 DUMMY_NODE_VECTOR0 Vector__XXX;
BA_DEF_ BU_  "NodeLayerModules" STRING ;
BA_DEF_ BO_  "GenMsgStartDelayTime" INT 0 65535;
BA_DEF_ SG_  "NWM-WakeupAllowed" ENUM  "No","Yes";
BA_DEF_ BO_  "NmMessage" ENUM  "no","yes";
BA_DEF_ BO_  "GenMsgILSupport" ENUM  "No","Yes";
BA_DEF_ BU_  "NmNode" ENUM  "no","yes";
BA_DEF_ BU_  "NmStationAddress" INT 0 63;
BA_DEF_  "NmBaseAddress" HEX 1024 1087;
BA_DEF_ BO_  "GenMsgCycleTimeFast" INT 0 50000;
BA_DEF_  "BusType" STRING ;
BA_DEF_ BO_  "GenMsgCycleTime" INT 0 50000;
BA_DEF_ BO_  "GenMsgDelayTime" INT 0 1000;
BA_DEF_ BO_  "GenMsgNrOfRepetition" INT 0 999999;
BA_DEF_ BO_  "GenMsgSendType" ENUM  "Cyclic","NotUsed","NotUsed","NotUsed","NotUsed","NotUsed","NotUsed","IfActive","NoMsgSendType","NotUsed","vector_leerstring";
BA_DEF_ SG_  "GenSigInactiveValue" INT 0 100000;
BA_DEF_ SG_  "GenSigSendType" ENUM  "Cyclic","OnWrite","OnWriteWithRepetition","OnChange","OnChangeWithRepetition","IfActive","IfActiveWithRepetition","NoSigSendType","NotUsed","NotUsed","NotUsed","NotUsed","NotUsed";
BA_DEF_ SG_  "GenSigStartValue" FLOAT 0 100000000000;
BA_DEF_DEF_  "NodeLayerModules" "CANoeILNLVector.dll";
BA_DEF_DEF_  "GenMsgStartDelayTime" 0;
BA_DEF_DEF_  "NWM-WakeupAllowed" "";
BA_DEF_DEF_  "NmMessage" "no";
BA_DEF_DEF_  "GenMsgILSupport" "Yes";
BA_DEF_DEF_  "NmNode" "no";
BA_DEF_DEF_  "NmStationAddress" 0;
BA_DEF_DEF_  "NmBaseAddress" 1024;
BA_DEF_DEF_  "GenMsgCycleTimeFast" 0;
BA_DEF_DEF_  "BusType" "";
BA_DEF_DEF_  "GenMsgCycleTime" 0;
BA_DEF_DEF_  "GenMsgDelayTime" 0;
BA_DEF_DEF_  "GenMsgNrOfRepetition" 0;
BA_DEF_DEF_  "GenMsgSendType" "NoMsgSendType";
BA_DEF_DEF_  "GenSigInactiveValue" 0;
BA_DEF_DEF_  "GenSigSendType" "Cyclic";
BA_DEF_DEF_  "GenSigStartValue" 0;
BA_ "GenMsgILSupport" BO_ 1 1;
BA_ "GenMsgSendType" BO_ 1 0;
BA_ "GenMsgCycleTime" BO_ 0 100;
BA_ "GenMsgILSupport" BO_ 0 0;
BA_ "GenMsgSendType" BO_ 0 0;

If you only have the definitions of the attributes and the default value everything is ok, but if you add a value to an attribute with ENUM type (BA_ ... is generated), that error pops up.

AndreiRadac, 2017-12-20:
@ebroecker : I just had a quick look. The conversion form DBC to xlsx is ok, but if I try to convert the generated xlsx back to DBC, this error pops up:

[image]

In the xlsx file I saw that now the Launch Type is between **" "**

[image]

ebroecker, 2017-12-21:
@AndreiRadac 
Did you try one more?

AndreiRadac, 2017-12-21:
@ebroecker 

I had a quick look today, and seems to be ok (at least what I tested). So I think you can close the issue.

closed 2017-12-21 by ebroecker