ebroecker/canmatrix #123: AttributeError:’Frame‘ object has no attribute 'Id'

orange2348, 2017-12-20:
hello
    I tried to use the canconvert tool to extract a ECU node from dbc, but when i run tool with command "canconvert --ecus KVM ***.dbc KVM.dbc". it reports 
"
Traceback (most recent call last):
  File "C:\Users\chenhechun\AppData\Local\Programs\Python\Python35\Scripts\canco
nvert-script.py", line 11, in <module>
    load_entry_point('canmatrix==0.6.1rc0', 'console_scripts', 'canconvert')()
  File "C:\Users\chenhechun\AppData\Local\Programs\Python\Python35\lib\site-pack
ages\canmatrix-0.6.1rc0-py3.5.egg\canmatrix\convert.py", line 353, in main
  File "C:\Users\chenhechun\AppData\Local\Programs\Python\Python35\lib\site-pack
ages\canmatrix-0.6.1rc0-py3.5.egg\canmatrix\convert.py", line 56, in convert
  File "C:\Users\chenhechun\AppData\Local\Programs\Python\Python35\lib\site-pack
ages\canmatrix-0.6.1rc0-py3.5.egg\canmatrix\copy.py", line 65, in copyBUwithFram
es
    __all__ = ["Error", "copy", "deepcopy"]
  File "C:\Users\chenhechun\AppData\Local\Programs\Python\Python35\lib\site-pack
ages\canmatrix-0.6.1rc0-py3.5.egg\canmatrix\copy.py", line 100, in copyFrame
    if reductor:
AttributeError: 'Frame' object has no attribute 'Id'
“

Do you know how to fix this fault?

Best regards!

ebroecker, 2017-12-20:
should be fixed with https://github.com/ebroecker/canmatrix/commit/0410988f83c30f079e47b97d29ae63ceb3966136 already

could you please retry current development-branch?

orange2348, 2017-12-27:
I tried the development-branch, it works. 
Thanks.

closed 2017-12-27 by orange2348