ebroecker/canmatrix #126: Scaling of min/max/default in signals

altendky, 2017-12-29:
Are the in-memory min/max/defaults supposed to be comparable to the raw integers in a CAN message?  or the human value they represent?  With a factor of 0.1 and a default human value of 15.1, should signal.attributes['GenSigStartValue'] be `'15.1'` or `'151'`?

This is coming up because some of my defaults are not getting written to my .sym file because after the scaling (introduced at the beginning https://github.com/ebroecker/canmatrix/commit/95611b1a8e4ebf22f14bd0ff08e348f66654c048#diff-2fda6f9ac5ab0214cac52642f17fc1f7R167) the value is not in the range of the min and max and so isn't output.  I just thought it would be good to know the intended in-memory value before trying to review the code.

For reference, here is a sample signal from the symbol editor.  Note that the max and default both have decimals so they are scaled values, not directly comparable to any raw integer value from a message.

```
Var=Var1 unsigned 0,8 /f:0.1 /min:11 /max:25.4 /d:15.1
```

ebroecker, 2017-12-29:
I know some dbc's where the default is value for a signal is 'invalid value' which is often not in range between min and max.
So maybe the min/max-check for the default-value is arguable.

In my opinion the raw-default-value should be used, because scaling often leads to some arithmetical errors. 

I think .dbc stores the raw-default value but .sym stores the scaled value?

altendky, 2018-01-02:
In addition to calculation errors, even just converting from a string to a float can induce error.  That is why I added the float factory option and use [decimal](https://docs.python.org/3/library/decimal.html) (https://github.com/ebroecker/canmatrix/commit/a622e5ebf2aac1fa226aea89474a24df6224c5b9).

But, not calculating is only an option if all file formats store all values either scaled or not.  Well, at least given the assumption that we want a consistent state in memory.  Alternatively we could calculate losslessly using decimal and/or [fractions](https://docs.python.org/3/library/fractions.html) as needed.  If some formats store scaled values and others raw/unscaled values then we are stuck doing calculations since the in-memory value should be consistent.

ebroecker, 2018-01-18:
... slowly but surely I understand this issue :-)
Well in my opinion fractions seem to be the best solution.

ebroecker, 2019-10-21:
A long time later, I am working on this issue.
I now decided to use the scaled values and store them  [decimal](https://docs.python.org/3/library/decimal.html) 

I have done this now for
```python
Signal.initial_value
```

closed 2019-11-08 by ebroecker